betaflight/betaflight #13824: OSD fails when GPS build options is used in the firmware build process

bhuism, 2024-08-15:
### Describe the bug

GPS and OSD can't be used together (I suspect something with the i2c bus since the OSD chip and the magnetometer are on the same but)

Regardless of connected GPS (i2c of serial) the OSD is unstable, is does not initialize (no OSD visible after boot), and when visible (after changes OSD items around) the fonts are garbled, and re-uploading the fonts does not help (als not with battery connected power up)

### To Reproduce

Only tested on "NeutronRC H743 bluetooth" board:

1. Compile firmware with GPS and OSD
2. Flash
3. Boot (with of without battery connected, with or without GPS connected)
3) OSD is unstable, not visible, can not upload fonts

I don't know if it is specific to this board or mcu

Only fix is not compiling GPS option in the firmware, then OSD is completely fine.

### Expected behavior

OSD and GPS  is fine and usable when both are compiled in the image.

### Support ID

```txt
b5db92e1-cd92-458b-8502-b3c4d6ceaa06
```

### Flight controller

NeutronRC H743 bluetooth

### Other components

VTX IRC Serial connected
4in1 ESC NeutronRC
(no RX)

### How are the different components wired up (including port information)

Problem is not related to wiring, just building the image with GPS and OSD created the problem

haslinghuis, 2024-08-18:
`GPS: NOT CONNECTED, UART2 115200 (set to 115200), NOT CONFIGURED, version =  unknown`

What GPS unit is used here - it does not seem to be recognized ?

Only MAG and BARO would use I2C in this configuration as only one is recognized (which guess would be BARO)

bhuism, 2024-08-22:
I will test it again, regardless of connected gps this happens

haslinghuis, 2024-08-22:
https://ae01.alicdn.com/kf/S8dfa2e96a67e4a098fa31e8df1d07f99I.png

haslinghuis, 2024-08-22:
Try in CLI adding

```
serial 1 2 115200 57600 0 115200
serial 2 1 19200 57600 0 115200
serial 3 8192 115200 57600 0 115200
serial 5 64 115200 57600 0 115200
serial 7 1024 115200 57600 0 115200
save
```

closed 2024-09-25 by bhuism