betaflight/betaflight #13837: Camera control doesn't work

NMFES, 2024-08-22:
### Describe the bug

Connected CC pad with OSD pin on the camera CaddX Ratel Pro (its pins are mixed up). Tried with default settings, HARDWARE/SOFTWARE mode.
Measured camera resistance - about 47kOhm, controls about default 450,270,150,68,0.
Tried with 150/300 Ohm resistor, with added capacitor (103 ceramic and 20mF).
Changed camera_control_key_delay  to upper/lower values. 
Checked [that list](https://www.rcgroups.com/forums/showthread.php?2961216-Betaflight-Camera-Control-Compatibility-Reports).

### To Reproduce

Solder a wire and check it :)

### Expected behavior

Expect camera control OSD appears using sticks

### Support ID

```txt
b3ad772d-8cdb-49ea-9d54-93fd63828cbc
```

### Flight controller

SpeedyBee F405 V4

### Add any other context about the problem that you think might be relevant here

Saw in Joshua's video, he said if the controller has CC pad then it already has a resistor connected to it. The wiring diagram also [says](https://store-fhxxhuiq8q.mybigcommerce.com/product_images/img_SpeedyBee_F405_V4_60A_Stack/SB_F405V460A-Other-2.jpg) it should work using soldered wire.

haslinghuis, 2024-08-22:
`CAMERA_CONTROL` has a timer defined but is not being assigned.

Try enabling `dshot_bitbang` and share output of `timer show`

NMFES, 2024-08-22:
> `CAMERA_CONTROL` has a timer defined but is not being assigned.
> 
> Try enabling `dshot_bitbang` and share output of `timer show`

Enabled dshot_bitbang.

# timer show

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

NMFES, 2024-08-22:
> Something is removing the timer assignment when applying your dump. This is after flashing

Changed to HARDWARE_PWM.

` timer show

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12:
    CH1 : CAMERA_CONTROL
TIM13: FREE
TIM14: FREE`

NMFES, 2024-08-22:
@haslinghuis Sorry, what should I do next?

nerdCopter, 2024-08-22:
> @haslinghuis Sorry, what should I do next?

i think set `camera_control_mode = HARDWARE_PWM` (default)

NMFES, 2024-08-22:
@haslinghuis Already changed and pasted the output above

haslinghuis, 2024-08-22:
Try `set camera_control_internal_resistance = 270`
Second try additional `set camera_control_button_resistance = 450,260,210,150,50`
Third try additional `set camera_control_ref_voltage = 324`

There are different settings for Ratel 2 but not the Pro version.

NMFES, 2024-08-22:
@haslinghuis The thing is Caddx Ratel Pro wiring at this picture is [wrong](https://www.reddit.com/r/Multicopter/comments/j18sci/caddx_ratel_osd_w_camera_control_how/). It has a green wire which is a ground, and the black one is OSD. For those who will have the same issue. 

I've applied:
`set camera_control_internal_resistance = 270`
And it helped. Next I changed resistance of the buttons. 

By default, DOWN button didn't work. So next should work well:
`set camera_control_button_resistance = 450,260,150,70,5`

And I disabled dshot_bitbang. Do I need it?
Thank you for your help

haslinghuis, 2024-08-22:
> @haslinghuis The thing is Caddx Ratel Pro wiring at this picture is [wrong](https://www.reddit.com/r/Multicopter/comments/j18sci/caddx_ratel_osd_w_camera_control_how/). It has a green wire which is a ground, and the black one is OSD. For those who will have the same issue.

> I've applied:
set camera_control_internal_resistance = 270
And it helped. Next I changed resistance of the buttons.

> By default, DOWN button didn't work. So next should work well:
set camera_control_button_resistance = 450,260,150,70,5

:heart: Thank you for the feedback and glad we found the right settings.

:information_source: @ot0tot could better explain the pro's and con's of dshot bitbang setting.

closed 2024-08-23 by NMFES