betaflight/betaflight #13846: SD Card Fatal Error in Betaflight 4.5.1

aserpik, 2024-08-24:
### Describe the bug

I am getting a "fatal error" under the onboard SD Card in the Blackbox tab, regardless of whether I have an SDcard inserted or not. The SD card i am using is a 32gb card formatted to fat32. I've used various pieces of software to format it but nothing has worked. I get "fatal error, reboot to retry" with the SD card inserted and also when the slot is empty .

### To Reproduce

Connect quad. Insert SD card (optional). Go to Blackbox tab. 

### Expected behavior

SD Card recognized in blackbox tab. 

### Support ID

```txt
# 

# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.5.1 Aug 21 2024 / 04:30:14 (77d01ba3b) MSP API: 1.46
# config rev: d69ac5f

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name JBF7_V2
manufacturer_id IFRC
mcu_id 0024003d5746501520303431
signature 

# feature
feature GPS
feature TELEMETRY
feature OSD

# serial
serial 0 131073 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 3 2 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 2 1350 1750 0 0
aux 2 2 2 1700 2100 0 0
aux 3 46 1 1700 2100 0 0
aux 4 13 5 1700 2100 0 0
aux 5 35 3 900 1300 0 0

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 0
set dyn_notch_count = 1
set dyn_notch_q = 700
set dyn_notch_min_hz = 150
set dyn_notch_max_hz = 650
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 775
set acc_calibration = 130,222,36,1
set align_mag = CW270FLIP
set mag_align_pitch = 1800
set mag_align_yaw = 2700
set min_check = 1010
set max_check = 1990
set rc_smoothing_auto_factor = 140
set rc_smoothing_auto_factor_throttle = 45
set rc_smoothing_setpoint_cutoff = 25
set rc_smoothing_feedforward_cutoff = 25
set blackbox_sample_rate = 1/2
set blackbox_mode = ALWAYS
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_output_reordering = 1,3,0,2,4,5,6,7
set failsafe_procedure = GPS-RESCUE
set mixer_type = EZLANDING
set small_angle = 180
set mag_declination = 110
set gps_sbas_mode = WAAS
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_allow_arming_without_fix = ON
set simplified_gyro_filter_multiplier = 155
set sdcard_detect_inverted = OFF
set sdcard_spi_bus = 1
set osd_warn_bitmask = 286717
set osd_link_quality_alarm = 40
set osd_rssi_pos = 1389
set osd_link_quality_pos = 3437
set osd_link_tx_power_pos = 3373
set osd_rssi_dbm_pos = 3405
set osd_flymode_pos = 2616
set osd_throttle_pos = 2341
set osd_crosshairs_pos = 2360
set osd_ah_sbar_pos = 2361
set osd_ah_pos = 2233
set osd_gps_speed_pos = 2196
set osd_gps_sats_pos = 2436
set osd_home_dir_pos = 2204
set osd_home_dist_pos = 2534
set osd_compass_bar_pos = 2549
set osd_altitude_pos = 2597
set osd_warnings_pos = 2484
set osd_sys_lq_pos = 3180
set osd_gps_sats_show_pdop = ON
set debug_mode = MAG_CALIB
set rpm_filter_weights = 100,0,80

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 0
set dterm_lpf1_dyn_max_hz = 0
set dterm_lpf1_dyn_expo = 0
set dterm_lpf1_static_hz = 108
set dterm_lpf2_static_hz = 217
set anti_gravity_gain = 120
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set p_pitch = 67
set i_pitch = 103
set d_pitch = 42
set f_pitch = 147
set p_roll = 61
set i_roll = 93
set d_roll = 37
set f_roll = 135
set p_yaw = 61
set i_yaw = 93
set f_yaw = 135
set d_min_roll = 37
set d_min_pitch = 42
set feedforward_smooth_factor = 30
set feedforward_jitter_factor = 12
set feedforward_boost = 10
set simplified_master_multiplier = 125
set simplified_i_gain = 85
set simplified_pi_gain = 110
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 90
set simplified_pitch_pi_gain = 105
set simplified_dterm_filter_multiplier = 145
set ez_landing_limit = 5

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 2
set pitch_rc_rate = 2
set yaw_rc_rate = 2
set roll_expo = 40
set pitch_expo = 40
set yaw_expo = 40
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 60

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#
```

### Flight controller

JBF7 v.2

haslinghuis, 2024-08-24:
`set sdcard_spi_bus = 1` is supposed to be 3

aserpik, 2024-08-24:
sdcard_spi_bus was originally set to 3. I changed it to 1 trying to troubleshoot the issue and forgot to change it back. Changing it back to 3 does not fix the problem unfortunantly.

haslinghuis, 2024-08-24:
- Does it work in former firmware?
- Need to format using https://www.sdcard.org/downloads/formatter/

aserpik, 2024-08-24:
The formatter you linked worked! Thank you!!

closed 2024-08-24 by aserpik