betaflight/betaflight #13847: MSP over CRSF - only 2 chunks are received for large MSP frames

JimB40, 2024-08-24:
### Describe the bug

When MSP frame with large payload  is requested ie
48 (0x30) MSP_LED_STRIP_CONFIG (length 130)
52 (0x43) MSP_ADJUSTMENTS_RANGES (ength 208)
52 (0x54) MSP_OSD_CONFIG (length 208)
and few others (71, 116, 127, 164)

only two chunks (CRSF frames) are sent back. in total 112 bytes of payload
Resulting in missing data. 

Tested on BF 4.4.x and 4.5.x 

### To Reproduce

Check incoming frames in Betaflight Setup LUA selecting "OSD Elements" that uses MSP_OSD_CONFIG command

### Expected behavior

all chunks should be sent back 

### Support ID

```txt
a1faa56f-ca73-4308-b7b7-36228f0d897d
```

### Flight controller

STM32F411

haslinghuis, 2024-08-24:
Is this a LUA or firmware (MSP) issue ?

JimB40, 2024-08-24:
firmware issue IMHO.
Used my MSP testing LUA and modified Betaflight Setup LUA to see on radio LCD what is the response from firmware.

There is still open #11928 about MSP_GPS_SV_INFO sending only 0 over CRSF until usb cable is connected and Betaflight Configurator opened. Looking at screenshot from tests I've done all 3 response chunks for MSP_GPS_SV_INFO (length 129) were sent back properly. It was tested on 4.3.x

JimB40, 2024-08-27:
Update about 4.5.1
At first I've seen only first chunk incoming, but by accident sent request again and next chunk came.

This is different behaviour from 4.3 and 4.4 where you send MSP request once and all chunks (CRSF frames for big MSP packet) are comming?
So my qustion is is that a bug or feature?

haslinghuis, 2024-08-27:
@klutvott123 ?

haslinghuis, 2024-09-06:
@ledvinap ?

ledvinap, 2024-09-08:
@JimB40 It's a bug, repeated requests are not working as intended. 
But I don't see any obvious cause of this, maybe it's something scheduler related.

Can you try bisecting the problem?
ca you try it with crsf_use_negotiated_baud = ON ?

klutvott123, 2024-09-12:
Is it ELRS? ELRS can't do an arbitrary number of frames. The limit is 2.
crsf_use_negotiated baud won't do anything

JimB40, 2024-09-16:
Some bigger MSP frames using CRSF need 3 or maybe more chunks.
Then what is desicion?
Should I leave in code additional request for every chunk?
Seems that newest BF Script works lke that.

ledvinap, 2024-09-16:
@JimB40 : Can you test if any MSP request will return another part of first reply packet?

JimB40, 2024-09-17:
Yes it will. 
Test with MSP_ADJUSTMENTS_RANGES (length 208)
[image]

1. Above green line
to get next chunk I send 0x34 request all the time

2. Below green line
for first chunk I send 0x34 request then for 2nd, 3rd & 4th I send 0x01 request (MSP_API_VERSION)
So it seems no mater what request command you send -> BF will continue to send next chunks from previous command until BF MSP out buffer is empty.
When empty it will respond with first chunk of actual command sent.

JimB40, 2024-10-08:
Any progress?

klutvott123, 2024-12-08:
> Any progress?

Seems like a ELRS issue. When sending a request all chunks should be sent back without having to send another request. This is the way it works for all other protocols that support MSP over telemetry.

JimB40, 2024-12-30:
> Is it ELRS? ELRS can't do an arbitrary number of frames. The limit is 2.
> crsf_use_negotiated baud won't do anything

So testing again on 4.5.1 using MSPv1
- S.Port - works properly all frames are sent back. although there many errors (missing or doubled  chunks so it takes time to get proprer MSP frame if payload is bigger)
- Ghost - as above
- CRSF on TBS - works properly all frames are sent back.
- CRSF on ELRS - two frames are sent back at once and only first two chunks (frames) so if payload is bigger EdgeTX never gets it. 

@klutvott123 you wrote limit is 2. Is this due to some ELRS limitation? I thought it uses the same CRSF protocol.

haslinghuis, 2025-06-13:
@JimB40 this seems to be resolved in https://github.com/ExpressLRS/ExpressLRS/pull/3229

closed 2025-06-13 by haslinghuis