betaflight/betaflight #13856: OSD not working for CrazybeeF4FR with betaflight 4.5.x

probstn, 2024-08-28:
### Describe the bug

OSD (Happymodel Moblite7 with  CrazybeeF4FR and TBS SmartAudio on UART2) does not work with Betaflight version 4.5.0 or newer.
The OSD simply doesn't show up, even though activated and configured. To fix the issue, I did a rollback to version 4.4.3 and it worked first try.
I tried various other versions and had no problems except for versions above 4.5.0, where the OSD simply doesn't appear and I just get the video feed.

### To Reproduce

Flash Betaflight >4.5.0 on your CrazybeeF4FR and try to get OSD to work.

### Expected behavior

OSD should show up

### Support ID

```txt
71bd847d-5fea-4ffd-98b6-d4aae5af549e
```

### Flight controller

CrazybeeF4FR

### Other components

Happymodel Moblite7 AIO

### How are the different components wired up (including port information)

default

nerdCopter, 2024-08-28:
according to the suport ID, you have flashed with both SD and HD support which will always default to HD( MSP Display) settings.
[image]

Consider flashing with only SD or use the CLI to set for SD (Analog).
[image]
Further Details: https://hackmd.io/@nerdCopter/r1JbnG0Q0

kedeng, 2024-09-12:
@probstn Can you try set video format to NTSC ?
[image]

I got similar issue,  if we set video format to AUTO, the MAX7456 will try to identify video format. But sometimes it fail. In BF 4.4.3 the video format will set to NTSC， in 4.5.x it will set to PAL:

[Change MAX7456 AUTO default to PAL #13235](https://github.com/betaflight/betaflight/pull/13235/commits/e55acf95fec061869c5e5c35aa64137002846eb0)

So you may see OSD work fine in 4.4.3 but not 4.5.x

closed 2024-10-21 by github-actions[bot]