betaflight/betaflight #13857: Smartaudio won't work between JHEF405 GF16 and Betafpv M03

G4E, 2024-08-29:
### Describe the bug

Hello,
I have a JHEF405 GF16 FC and a betafpv M03 VTX, and I can't seems to make smartaudio working between the two. This is somewhat problematic as this vtx don't have other means to change frequency.

I connected the VTX on the UART4 TX pad as described by the manufacturer :
[image]

However I always have "Device Ready : False" in the vtx tab.

The VTX is powered with a 5V pad which is **not** from USB.

I captured the traces with my oscilloscope and I can't make sens of what I see :
When only the FC is powered by USB, the signal seems ok :
[image]

However when I powered the whole drone (thus the VTX) from the battery, the signal seems "squished" on the upper level :
[image]

I tried adding a 50KO pull down with no effect.

Thank you for your time,
G4E.

### To Reproduce

* Connect M03 VTX SA to JHEF405 TX4
* Enable SmartAudio in configurator
* The status is stuck on "Device Ready : False"

### Expected behavior

I expect communication between M03 and JHEF405

### Support ID

```txt
# ###

# # Problem description

# # No smartaudio between betafpv M03 and JHEF405

# ###

# version
# Betaflight / STM32F405 (S405) 4.5.1 Aug  5 2024 / 10:41:40 (77d01ba3b) MSP API: 1.46
# config rev: ad0057e
# board: manufacturer_id: JHEF, board_name: JHEF405PRO

# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.29V, Core temp=52degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 3789, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=ICM42688P, ACC=ICM42688P, BARO=DPS310
OSD: MAX7456 (30 x 13)
BUILD KEY: 38618efabf1fa57194bc410eb52471bb (4.5.1)
System Uptime: 236 seconds, Current Time: 2024-08-29T11:00:42.276+00:00
CPU:43%, cycle time: 123, GYRO rate: 8130, RX rate: 15, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
FLASH: JEDEC ID=0x00ef4018 16M
GPS: NOT ENABLED
Arming disable flags: RXLOSS CLI MSP

# flash_info
Flash sectors=256, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=16777216 JEDEC ID=0x00ef4018
Partitions:
[... 1113 lines cut ...]
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

*** Support data submitted *** 
Id: "8bc772b1-dd4a-4814-9879-b4289679da95"

# copy ID and provide to the betaflight team.
```

### Flight controller

JHEF405 GF16

### Other components

* ELRS RX
* M10 GPS
However those were not powered during the capture with oscilloscope

### How are the different components wired up (including port information)

* ELRS on UART1
* GPS on UART3

haslinghuis, 2024-08-29:
Please test https://github.com/betaflight/betaflight/pull/13797

haslinghuis, 2024-08-29:
Please test https://github.com/betaflight/betaflight/pull/13797#issuecomment-2269376930

G4E, 2024-08-29:
@haslinghuis 
The issue is exactly the same with #13797 
[image]

@ledvinap 
Yes, I check with oscilloscope with and without pulldown, and HI voltage is the same in both cases

G4E, 2024-08-29:
I have to add that I bought this VTX used, and I didn't check if SA worked on another FC yet. Except this issue, it seems fully fonctionnal (the button change the LED color & power, and it emits video correctly)

ledvinap, 2024-08-29:
@G4E : Can you re-check wiring?

- check that there is no voltage between FC gnd and VTX gnd (both with VTX powered and unpowered). Use some decet GND pad on VTX (or U-fl shielding), not the one used t connect to FC
- measure VTX voltage (ideally regulated one)
- (if easy) - measure voltage on VTX input with 50kOhm to 3.3V and 0V (pullup/pulldowd)

G4E, 2024-08-29:
> @G4E : Can you re-check wiring?
>
>     * check that there is no voltage between FC gnd and VTX gnd (both with VTX powered and unpowered). Use some decet GND pad on VTX (or U-fl shielding), not the one [...]

1. 0v between UFL shield and FC GND
2. 5.01 Volts
3. without pullup nor pulldown , SA is at 3.5V. With pulldown SA is also at 3.5V. I can't measure with pull up sorry.

However I tried downgrading to 4.4.2 and 4.3.2 and the exact same issue remains. I also tried to use softserial, and the result is the same.
I think it is safe to assume this is not a 4.5 issue....
Feel free to close the issue, but i'm not sure what is wrong here, any help is appreciated

ledvinap, 2024-08-29:
> without pullup nor pulldown , SA is at 3.5V. With pulldown SA is also at 3.5V. I can't measure with pull up sorry.

Without FC connected? In that case, the VTX pin is probably fried ..

G4E, 2024-08-29:
I swapped the M03 with a TX805 and it was detected first try, no issue whatsoever.
This VTX is fried. Sorry for bothering you, and thanks for your help !

closed 2024-08-29 by G4E