betaflight/betaflight #13859: Uncommanded Ascending on Zero Throttle from Airmode yaw wind-up

AfterFPV, 2024-08-29:
### Describe the bug

While dropping Throttle to Zero my quadcopter continues to ascend.  Blackbox logs show that Motors 3 and 2 are start at 45% while 4 and 1 are at 0% while Airmode attempts to hold Yaw steady.  Some sort of wind-up occurs and gets stronger until 3 and 2 are at 70-75% causing a Zero throttle quadcopter to ascend.  To avoid flyaway I physically tip the quadcopter upside down to actually descend.  Very scary and dangerours this has occured while attempting to descend around manned aircraft.
[image]

### To Reproduce

Drop throttle to Zero and descend with props facing up.

### Expected behavior

Expect quadcopter to descend.

### Support ID

```txt
###

# # Problem description

# # Uncommanded Ascending

# ###

# version
# Betaflight / STM32H743 (SH74) 4.3.1 Jul 13 2022 / 03:37:48 (8d4f005) MSP API: 1.44
# config: manufacturer_id: HBRO, board_name: KAKUTEH7, version: e4b51680, date: 2021-10-04T00:52:01Z
# board: manufacturer_id: HBRO, board_name: KAKUTEH7

# status
MCU H743 (Rev.V) Clock=480MHz, Vref=3.32V, Core temp=61degC
Stack size: 2048, Stack address: 0x2001fff8
Configuration: CONFIGURED, size: 4184, max available: 131072
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
OSD: MSP
System Uptime: 47 seconds, Current Time: 2024-08-29T22:03:28.676+00:00
CPU:17%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 10
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 9
SD card: Manufacturer 0x3, 15558144kB, 12/2016, v8.0, 'SS16G'
Filesystem: Ready
Arming disable flags: RXLOSS CLI MSP RPMFILTER

# flash_info
Flash sectors=0, sectorSize=0, pagesPerSector=0, pageSize=0, totalSize=0
Paritions:
FlashFS size=0, usedSize=0

# dump master

# version
# Betaflight / STM32H743 (SH74) 4.3.1 Jul 13 2022 / 03:37:48 (8d4f005) MSP API: 1.44
# config: manufacturer_id: HBRO, board_name: KAKUTEH7, version: e4b51680, date: 2021-10-04T00:52:01Z

# start the command batch
batch start

board_name KAKUTEH7
manufacturer_id HBRO

# name: Blastr6s-HD3

# resources
resource BEEPER 1 C13
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 B03
resource MOTOR 4 B10
[... 1243 lines cut ...]
```

### Flight controller

Kakute H7

### Other components

Frame | Xhover Blastr 5” Cinematic Frame
Motors | Brotherhobby Returner 6 2306 1800Kv
FC | Holybro Kakute H7
ESC | Tekko32 F3 Metal 65A 4in1
Camera | DJI Camera (Camera: 27.4 x 21.1 x 20.1mm)
vTx | Caddx Vista
vTx Antenna |  
Rx | TBS Crossfire Diversity Nano Rx
Rx Antenna(s) | Immortal T, Bardpole
Buzzer | Vifly Finder 2
GPS | TBS M8 GPS
Props | Gemfan Hurricane 51433 
Batteries | Tattu R-Line 4.0 1050

### Add any other context about the problem that you think might be relevant here

I have 3 quadcopters with Kakute flight controllers and identical firmware versions.  Only one quadcopter exhibits this behavior while not being noticeably different in weight or power.  I've double checked the configs are the same.  Only unique thing about it is it's a True-X frame.  I've reset the firmware, tried different props and tunes, still only this quad has issues.

AfterFPV, 2024-08-29:
https://youtu.be/omm1p_UMfRg

ctzsnooze, 2024-08-30:
This has been reported variously in different places.
Yaw is strongly dependent on motor differentials - two motors must speed up, and two others must slow down, to yaw.  So if there is a persisting reason for yaw PIDs to 'maintain a yaw', there will be an effective increase in throttle.  This is the cause of the transient 'yaw jump' behaviour.  It's also the cause of what you see.
A fast yaw change is implemented by rapidly changing the speed of the motor pairs, and the rotatory momentum change due to motor acceleration immediately effects a rotation of the quad.  This doesn't need iTerm buildup. This effect is much weaker on large quads than small quads.
A slow and persistent yaw request, eg a user requesting a long slow left yaw, means that we want a constant yaw over a long time.  This cannot be achieved by accelerating or decelerating the motors, and is achieved by 'drag' on the air by constant motor differential.  Larger quads do this better than smaller quads.  This 'buildup' to the required constant motor differential is mediated by iTerm.  A constant motor differential of say 50% means that one pair of motors will be 50% and the others 0%, ie an effective throttle of 25%.  If iTerm can build up on yaw to take 100% of throttle, then you can conceivably get up to 50% net throttle and an ongoing climb.

This is what is happening to you.

There are several ways to avoid this.
- Don't fly long sustained yaw inputs that cause iTerm to grow
- Check that your motor's KV are exactly the same, otherwise sometimes there is a net yaw effect that gets concealed but results in a yaw iTerm accumulation in normal flight
- Check that there is no aerodynamic reason for a net yaw drag effect
To check for points 2 and 3, fly in normal straight forward flight and check that iTerm on yaw is zero or close to zero.

Setting yaw iTerm values lower than default will make it take proportionally longer for the problem to accumulate, but also a lot longer for it to go away.

The software mitigation is to not allow yaw, or more explicitly, not allow yaw iTerm, to command too much of the available motor mix.  This will result in failure to achieve long yaw sustained yaw inputs once the iTerm reaches its limit.

There are improvements in how this is done since 4.3.

In 4.6 we have improved feedforward on yaw explicitly so that less iTerm is built up on transient moves. That won't help your problem, but reduces transient iTerm growth, and allows lower iTerm values overall.

We also have https://github.com/betaflight/betaflight/pull/13543 which changes how we limit iTerm, by improving how `iterm_windup` works. It allows you to use `iterm_windup` to limit the percentage of the overall `pidsum_limit_yaw` value.  On the defaults in that PR the yaw PIDsum is limited to 400, and with iterm_windup at 80%, yawiTerm cannot accumulate more than 320.  320 is still a lot; it will result in a 60% motor differential approximately, and could cause a climb.  So you could in this quad reduce the iterm_windup value to say 50%, limiting it to only 200, which means no more than 40% motor differential.

That PR provides a better tool for managing this problem than we have ever had before.

I'm not sure what the iTerm limit options are in 4.3, but basically you'd want to set it so that iTerm on yaw cannot get strong control over the motors.

In the meantime try to figure out why on this build yaw iTerm is building up as much as it does.  And avoid performing long duration low yaw rate arcs, if possible, since they will inevitably result in yaw iTerm building up.

Some builds with a forward GoPro have issues when yawing slowly at speed because the forward aerodynamics cause fundamental instability - imaging having a Fin at the front of the craft - then the PIDs have to push all the time against the aerodynamics during a sustained yaw.  To check that this is not the case on this build, remove the entire GoPro mount and repeat a test which previously caused motor windup.

Finally if you have, for instance, done a long sustained left yaw, and then got motor windup, initiate a substantial reverse yaw input for several seconds at zero throttle until the motors slow down.  That will take out the yaw iTerm that you accumulated, and the motors will calm down.

nerdCopter, 2024-08-30:
maybe share the BBL as well.

AfterFPV, 2024-08-30:
[LOG00001 - Copy.BFL.txt](https://github.com/user-attachments/files/16822611/LOG00001.-.Copy.BFL.txt)

~ 41 seconds excessive windup occurs at 0 throttle 0 yaw input.  From overexposed video attached above.

AfterFPV, 2024-08-30:
@ctzsnooze Thanks for the write-up and links, I think you've correctly identified the issue.  A few questions/clarifications:

- Explicitly this is Iterm windup not an Airmode issue?  If I disable airmode the flyaway problem goes away, but flails at 0 throttle 0 yaw.  My understanding of airmode was it'd raise all 4 motors up 15% allowing +/- 15% yaw control, but the logs paint a different picture going up to 75%.
- Unfortunately I can't fly around this, every cliff dive involves dropping throttle to zero and little to no yaw input, but only this one quad flys off at 50% throttle while I'm inputting none.  I've learned to wiggle my nose or flip but it ruins the fun/shot.  Example of identical firmware quads behaving very differently: https://youtu.be/uQuCIjah1UA
- I believe the "aerodynamic" reason makes the most sense.  The True-X frame, bardpole, and far forward gopro could all influence that.  The right pictured quad has the issue whereas left flies fine.
[image]

- I'm open to reverting to 4.2 if that'd help, upgrading to 4.5, waiting for 4.6?  I can also continue limited testing on 4.3 if that's helpful, although I'm at the point of feeling this build is too dangerous to fly.
- I've checked all motors are same, but am considering swapping out motors and frame hoping this goes away, thoughts?

Summary: At zero throttle and zero yaw input my quad aerodynamically lists to the left,  I-term raises to compensate keeping quad facing forwards.  However at 15% quad appears to successfully hold yaw, why "windup" over several seconds to 100% if the 15% is working?

KrissV2, 2024-08-31:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

Quick-Flash, 2024-09-01:
This is a mechanical issue with your drone. Some aerodynamic forces are your drone are causing yaw issues. The thing is that if yaw was doing something dumb you would see the drone doing a good bit of yawing, its not. The yaw pid controller is holding yaw pretty straight. In order to do that it needs to command a fairly high throttle. The only software solution is to allow for the drone to spin on the yaw axis and to lower throttle, or to hold yaw straight and have throttle set fairly high.

You could of course disable airmode to get the drone to command less yaw and thus less total throttle, but the yaw would start drifting in this situation. So unfortunately their needs to be some sort of hardware change to change the aerodynamics of your drone. Maybe some motors are tilted slightly, or perhaps switching to some beefier props that have more yaw torque would help.

AfterFPV, 2024-09-04:
I’ve run the tests requested.  To me it looks like the forward mounted GoPro is the cause, but if you could look at the blackboxs/videos I’d really value your input.

- blastr 5” without GoPro 8 - 
[LOG00001-DVR_Only - Copy.BFL.txt](https://github.com/user-attachments/files/16880452/LOG00001-DVR_Only.-.Copy.BFL.txt)
- blastr 5” with GoPro 8 -   
[LOG00002-GoPro - Copy.BFL.txt](https://github.com/user-attachments/files/16880455/LOG00002-GoPro.-.Copy.BFL.txt) - Issues begin ~1:12

I can easily recreate the iterm windup with the GoPro, whereas with no GoPro I cannot.  Short video comparison(https://youtu.be/KyKdoR8RBoA).  If ‘I’ still seems high without GoPro, please let me know, I suspect GoPro alone isn’t the only cause.

This same quadcopter flew Betaflight 4.2.3 with Gopro for 2 years with no issues around iterm windup.  I made the following hardware and software changes, I suspect one or a combination of these changes is causing my flyaways.

- Swapped FC from Kakute F7 -> Kakute H7
- BF 4.2.3 -> 4.3.1 (4.3.1 dump attached above and 4.2.3, both preset or stock PIDS/filters)
[blastr6s.dump.txt](https://github.com/user-attachments/files/16880555/blastr6s.dump.txt)
- Swapped ImmersionRC 5.8 Ghz VTX for Caddx Vista (heavier, mounted in rear)
- Swapped Immortal-T for Bardpole (long fin mounted in rear)
- Changes to TPU mounts on rear
- Lowered prop pitch 51477 -> 51433
- Swapped front right motor for same motor, this quads had enough crashes any of the four could be damaged/underperforming

I’d like to address root cause and limit yaw iterm authority long term.  I’ve looked at the PR ( #[13790](https://github.com/betaflight/betaflight/issues/13790)), apologies for my surface level understanding of the PID loop, but it looks promising.  I’d be open to cherry-picking that in and testing.  I can’t downgrade to 4.2.3 via the configurator but could try my config files with 4.2.11 if that’s useful as well?  I'll also look at re-designing the GoPro mount to make it more centered.

ctzsnooze, 2024-09-04:
That PR has just been merged, so you can limit iTerm on yaw very easily now.
As I explained previously, there's no need to swap FC's or use old firmware or any of that stuff.  The cause is multiple changes to iTerm on yaw, and iTerm limiting, over several years.  And in your quad, the cause is having a large aerodynamic object at the front, which results in ongoing yaw error and yaw iTerm accumulation due to adverse aerodynamic design of the build.

In master (nightly) code now we have now:
- improved yaw feedforward, so you won't need so much yaw iTerm.  In fact yaw I can be cut to about half.
- made it so that iterm_windup really does stop iTerm from accumulating beyond the set percentage
- applied the iterm_windup percentage to the new pidsum_limit_yaw value, and 
- defaulted pidsum_limit_yaw to 400 and iterm_windup to 80.

With these settings, iTerm on yaw cannot exceed 320, or about 64% throttle to the two fast motors with idle on the other two. That's still high, but it's not as high as before.  

For quads without aerodynamic issues this should be OK, but there are almost no situations where anyone would really want iTerm on yaw of 320, not really.  So it could be that we should lower the iTerm_windup percentage to 50 by default.  Then the max iTerm on yaw would be 200.

To understand pidSUm values, a pidSum of 500 will fully saturate motorMix, ie a yaw pidSUm of 500 will set two motors to max, and two to zero/idle, at zero throttle.

A yaw pidSUm of 200 will send two motors to 40% and two to zero/idle, at zero throttle.  That's the equivalent of 20% throttle when iTerm is fully loaded up.  

So long as your quad won't climb at 20-25% throttle, a setting of 50% for iTerm_windup with default pidsum_limit_yaw of 400 will prevent the quad from climbing no matter how much the quad wants to increase yaw iTerm.  What will happen in flight is that the yaw capability of the quad will be limited; it will reach some maximum yaw rate in sustained yaw requests, due to the aerodynamic problem you have, and then it won't achieve the requested yaw rate; the yaw arc will open up a bit.  

Please test with master.
- Cut your pid iTerm value for yaw to 40 from default of 80 (will need to disable sliders on yaw by choosing RP slider mode).  That will make it take twice as long for the original problem to evolve.
- Set iterm_windup to 50 (50%)
- check that pidsum_limit_yaw is 400.

With this your max effective 'extra throttle' from fully loaded yaw iTerm is 20%.  

Check it doesn't climb uncontrollably during sustained yaw arcs.  You may find that to maintain constant altitude you have to cut throttle a bit, it would need to be at or below 20%

You'll find some iterm_windup value where it won't climb; the highest value that never causes a climb problem will give you the best sustained yaw performance.

AfterFPV, 2024-09-05:
That’s great news, thanks!  I’ve installed 4.6.0-dev from master, ported my settings where possible, and left default filters and PIDS.  I then followed your instructions:
- Set Yaw ‘I’ to 40
- Set iterm_windup to 50
- Verified pidsum_limit_yaw is 400

I’ve left the aerodynamics as is to get a more accurate comparison to the last flights.  Attached is the 4.6 config, let me know if there’s a debug mode or any further changes that could be useful.  I’ll report back with blackbox logs!

[Blastr6s-HD46.dump.txt](https://github.com/user-attachments/files/16899530/Blastr6s-HD46.dump.txt)

Quick-Flash, 2024-09-05:
> With this your max effective 'extra throttle' from fully loaded yaw iTerm is 20%.
>
> Check it doesn't climb uncontrollably during sustained yaw arcs. You may find that to maintain constant altitude [...]

If this prevents any yaw related fly aways, then yaw performance will suffer. The drone will yaw when not asked to. Currently the drone is doing a pretty good job of holding yaw, so reducing yaw authority will mean that yaw does indeed now deviate and slide around.

Quick-Flash, 2024-09-05:
> This same quadcopter flew Betaflight 4.2.3 with Gopro for 2 years with no issues around iterm windup. I made the following hardware and software changes, I suspect one or a combination of these [...]

Its likely changing your propellers to a lower pitch prop. That would decrease your yaw authority and I think is the most likely culprit.

AfterFPV, 2024-09-06:
Here's 4.6 with requested settings reflying same route with gopro, unable to recreate issue.
[LOG00006.BFL.txt](https://github.com/user-attachments/files/16901547/LOG00006.BFL.txt)

I had extra battery so redid the run with Yaw 'I' reset to default of 80, still unable to recreate issue.
[LOG00007.BFL.txt](https://github.com/user-attachments/files/16901548/LOG00007.BFL.txt)

Quick-Flash, 2024-09-06:
> Here's 4.6 with requested settings reflying same route with gopro, unable to recreate issue. [LOG00006.BFL.txt](https://github.com/user-attachments/files/16901547/LOG00006.BFL.txt)
>
> I had extra [...]

Huh after looking through the logs it appears that I am wrong. Apparently you can fix this without creating any real yaw issues. If your yaw performance is worse now its minimally worse, something you probably couldn't even notice.
@ctzsnooze good work!

haslinghuis, 2024-09-06:
Fixed in https://github.com/betaflight/betaflight/issues/13790

closed 2024-09-06 by haslinghuis