betaflight/betaflight #13861: Roll & pitch expo 2d vector calculation to correctly maintain user's intended direction and magnitude

gratif3, 2024-08-30:
### Is your feature request related to a problem? Please describe

The current math calculating expo on pitch and roll is 'incorrect' because it does not preserve the user's intended direction, and also does not have a consistent magnitude as you sweep around the circle. In flight it's a somewhat subtle feel difference, but this proposed way feels cleaner and more consistent. The math issue is essentially that currently a non-linear function ('expo') is being applied to both components independently of the user's input vector (pitch & roll stick). Because this is non-linear, the vector is now pointing in a slightly different direction (several degrees) and has a different magnitude even for user input vectors of the same original magnitudes. The correction is to apply expo to the *length* of the user's pitch & roll vector, then to break out the two components of that for the rest of the flight control code to run with.

details here: https://github.com/betaflight/betaflight/discussions/13836

### Describe the solution you'd like

https://github.com/gratif3/betaflight/commit/2d2815a4dbeeee644b68b6d2f30214d94740f928

I've implemented the solution here.

### Describe alternatives you've considered

Need to decide how to handle the (rare?) case when the user wants different pitch and roll rates, currently it only uses pitch rates for both axes. Also need to consider how to let the user select this or disable it. Could be a checkbox in the configurator near the expo settings.

### Other information

Here's a chart showing a vector field where the dark arrows and lines are the current method and the lighter arrows and lines are the proposed method. [image]

Quick-Flash, 2024-09-01:
https://www.desmos.com/calculator/rsnyrvgssi
Here I have graphed the effect. If you hold roll steady but continue to increase pitch, then the roll value will also continue to increase. It will do so in this way where eventually when you hit the circle of your gimbal (the point where if you drew a perfect circle your controller would touch it) that the axis with less movement will act as though it has no expo. The behaviour after that point is also odd as previously more pitch was leading to more roll (essentially removing expo from roll), but after your leave the circle area roll will begin to decrease as you add pitch. You should probably try to do something to smooth that effect out a bit, but this would be kind of odd to use in practice, but it might be pretty neat as well. I'm not sure how it would feel, and I do personally use a fair bit of expo, so this would likely change my stick feel even more, but I'm keen on trying it. I'm not sure that your analysis of why it is needed makes a ton of sense to me, but hey it might feel good regardless.

Quick-Flash, 2024-09-01:
If this is added it definitely needs to be optional as users are already use to the current way that things feel and I imagine many of them would not want to switch.

pdoersch, 2024-09-05:
Agreed on making this optional. FYI I did cap the input to only allow inside the unit circle since outside the circle doesn't really make sense to me. How can I make it easy for you to test? My commit is linked above if you want to build it. Or I could post builds for various configurations.

gratif3, 2024-09-05:
Agreed on making this optional. FYI I did cap the input to only allow inside the unit circle since outside the circle doesn't really make sense to me. How can I make it easy for you to test? My commit is linked above if you want to build it. Or I could post builds for various configurations. @Quick-Flash

gratif3, 2024-09-05:
Here are Unified Targets of https://github.com/gratif3/betaflight/commit/2d2815a4dbeeee644b68b6d2f30214d94740f928 to test:

[betaflight_4.5.1_AT32F435G.zip](https://github.com/user-attachments/files/16901143/betaflight_4.5.1_AT32F435G.zip)
[betaflight_4.5.1_AT32F435M.zip](https://github.com/user-attachments/files/16901144/betaflight_4.5.1_AT32F435M.zip)
[betaflight_4.5.1_STM32F7X2.zip](https://github.com/user-attachments/files/16901145/betaflight_4.5.1_STM32F7X2.zip)
[betaflight_4.5.1_STM32F405.zip](https://github.com/user-attachments/files/16901146/betaflight_4.5.1_STM32F405.zip)
[betaflight_4.5.1_STM32F411.zip](https://github.com/user-attachments/files/16901147/betaflight_4.5.1_STM32F411.zip)
[betaflight_4.5.1_STM32F446.zip](https://github.com/user-attachments/files/16901148/betaflight_4.5.1_STM32F446.zip)
[betaflight_4.5.1_STM32F745.zip](https://github.com/user-attachments/files/16901149/betaflight_4.5.1_STM32F745.zip)
[betaflight_4.5.1_STM32G47X.zip](https://github.com/user-attachments/files/16901150/betaflight_4.5.1_STM32G47X.zip)
[betaflight_4.5.1_STM32H723.zip](https://github.com/user-attachments/files/16901151/betaflight_4.5.1_STM32H723.zip)
[betaflight_4.5.1_STM32H725.zip](https://github.com/user-attachments/files/16901152/betaflight_4.5.1_STM32H725.zip)
[betaflight_4.5.1_STM32H730.zip](https://github.com/user-attachments/files/16901153/betaflight_4.5.1_STM32H730.zip)
[betaflight_4.5.1_STM32H743.zip](https://github.com/user-attachments/files/16901154/betaflight_4.5.1_STM32H743.zip)
[betaflight_4.5.1_STM32H750.zip](https://github.com/user-attachments/files/16901155/betaflight_4.5.1_STM32H750.zip)

Quick-Flash, 2024-09-05:
> Agreed on making this optional. FYI I did cap the input to only allow inside the unit circle since outside the circle doesn't really make sense to me. How can I make it easy for you to test? My [...]

If you want I could toss a few things together and you could choose which to use in the CLI.

nerdCopter, 2024-09-06:
> Here are Unified Targets of [gratif3@2d2815a](https://github.com/gratif3/betaflight/commit/2d2815a4dbeeee644b68b6d2f30214d94740f928) to test:

Proper way to offer this to public is make a pull request from a branch (not `master`), then pilots can cloud-build the PR. 
However, make it an option as discussed above.

ctzsnooze, 2024-09-07:
Yes, anyone can test this PR by including the PR with their online build.  That's the correct way to test and trial your PR.

gratif3, 2024-09-07:
Ok thanks for the interest! I made a pull request where this feature is enabled with a USE_VECTOR_EXPO #define at compile time.

closed 2026-09-26 by blckmn