betaflight/betaflight #13868: Save/load individual page configuration files

laegion420, 2024-09-02:
### Is your feature request related to a problem? Please describe

No, it's a feature request

### Describe the solution you'd like

A "save/export" and "load" button on each page in Betaflight to load and save pages separately to make it easier to apply certain settings over multiple drones or on new versions of betaflight. i.e. OSD layout, ports tab, configuration, modes. Like the vtx tab all could be easily saved and loaded individually from the corresponding tab.

### Describe alternatives you've considered

Saving the full config only works if it's the same drone, or same model.

### Other information

For instance, all of my GPS drones use the same OSD layouts, all of my non-gps drones use the same OSD layout. It's the same with modes, all of my GPS drones use the same mode/swith setup, all of them use the same receiver setup. This could be very useful for anyone running fleets of similar but not exactly the same drone.

nerdCopter, 2024-09-02:
easy to separate only the pertinent CLI diff/dump sections into preference `.txt` files and apply at will.

laegion420, 2024-09-02:
I understand that; but it's not intuitive or user-friendly. If it's already a function I'd think adding a few buttons that are programmed with CLI commands would not be extremely difficult to implement? Anyway, I'm not a dev, it's just  request that would be a feature that would help me out, and probably a lot of other people including people new to the hobby that are likely to be flying pre-built drones with a stock tune.

closed 2024-10-12 by github-actions[bot]