betaflight/betaflight #13870: Blackbox and GPS rescue issue on Mamba H743 V2

Dextyy1, 2024-09-02:
### Describe the bug

Good morning,

I have some issue after going to betaflight 4.5.0 recently with my drone. 

The main problem is that I can't use GPS rescue, even if the GPS is perfectly working. When I try to activate it in the "Failsafe" tab, it just goes back to "fall" after saving the configuration. Also, the GPS rescue mode is not displayed on the "Modes" tab.  
This is strange because the GPS is working perfectly, and during flight I have very accurate speed and home position.

My second issue is that blackbox is saving corrupted files. I never used it before, so I don't know if it is related to the current betaflight version, but each time I try opening logs files, I get an error on every software used. I can easily see in my files a bunch of strange characters after the configuration (see the example attached).

[btfl_001.zip](https://github.com/user-attachments/files/16841368/btfl_001.zip)

I didn't see any other issue with this version. Also for information I just tried flashing the 4.5.1 version, but the issues are still there.

Do you have any idea of what can cause this issues, and how to solve this ?

Regards.

### To Reproduce

I installed betaflight 4.5 on my quad. I didn't do anything in particular to have this issues.

### Expected behavior

GPS rescue should be bale to work with the current configuration, since it was working on betaflight 4.3. The only things that have upgraded since then is the GPS, because I put a more powerful one, which is working perfectly.
For blackbox, I don't know why I have corrupted files. I put a standard configuration of "Onboard flash", frequency 2kHz, mode "GYRO_SCALED", with all data included. I delete backlogs before each of my tests flights, and exporting the files right after each one by using "activate mass storage device mode".

### Support ID

```txt
247504b8-b0b9-4e62-beb0-34da9ce17745
```

### Flight controller

Mamba H743 V2 from Banggood

### Other components

- TBS crossfire nano RX
- Rush tank ultimate II VTX
- GPS Foxeer M10Q-250
- 4 in 1 55A Mamba ESC
- Autonomous beeper
- 4 WS2812B leds strip on each arm

### How are the different components wired up (including port information)

Direct wiring, with none of them going near the FC. All the component have their wire in a independent tube.
Port 1 : RX
Port 2 : GPS
Port 3 : VTX
Port 4 : Wifi
Port 6 : Camera control

### Add any other context about the problem that you think might be relevant here

FC : Mamba H543 Wifi (Dual Gyro)
Firmware : 4.5.0 BTFL
Betaflight Target : DIAT/MAMBAH743_2022B(STM32H743)
4 motors in bi-directionnal DSHOT600, for 3D flight

nerdCopter, 2024-09-03:
reference: https://github.com/betaflight/blackbox-log-viewer/issues/770

ctzsnooze, 2024-09-04:
First check the STATUS command in CLI, to ensure the GPS is getting auto-configured properly.
Then go looking for GPS_RESCUE commands with get GPS_RESCUE in the CLI, to check that they are all there.
Try setting the failsafe mode in the CLI, not in the Configurator GUI.

haslinghuis, 2024-09-05:
Feature 3D is enabled.

Dextyy1, 2024-09-05:
About blackbox issue, I try several solution, and I was able to get one well saved by disabling some of the option in blackbox. I need to try a little more, but maybe it was just because I was writing too much data at the same time.
About GPS_RESCUE, I have 3D mode on my quad, and after disabling it, I can have gps rescue, so thanks to you !
But what I can see in betaflight's guide, I should be able to have it, but it will just not work if I'm in 3D mode when failsafe is activated : 
"3D mode is not supported! There will be no rescue if the quad is in 3D mode at the time of the failsafe."
Maybe this is just not clear, at least now I know what was the issue, I think I will have to do a choice then.
I will try investigate if I can make blackbox work for something useful, thanks for your help !

haslinghuis, 2024-09-05:
Seems we have `DEBUG_BLACKBOX_OUTPUT`.

[image]

nerdCopter, 2024-09-06:
@ctzsnooze , i don't 3D nor have i any GPS experience so pardon if i'm missing something. That said, in regard to code, if 3D is switchable via AUX, then why not switch out of 3D, then perform the normal routines for Rescue/RTH when triggered?

closed 2024-10-15 by github-actions[bot]