betaflight/betaflight #13875: Garbage in the Craft name field and it's vars and parameters

SeriouSerg, 2024-09-04:
### Describe the bug

When you turn on GPS, the craft name parameter is overwritten. Moreover, all this is stored in EEPROM. 
I don’t know, maybe it occurred to someone to do this on purpose, but it was a very bad idea. I didn’t find a way to cut this out in the source code. Please remove it, or explain why you had to put all this crap in the craft name.

USE_CRAFTNAME_MESSAGES is commented.

### To Reproduce

1. Put something into craftName parameter
2. Reboot and enable GPS
3. Enjoy

### Expected behavior

Do not write anything in this field. If someone needs a status line, make a separate one.

### Support ID

```txt
not needed
```

### Flight controller

MATEKH743

### Add any other context about the problem that you think might be relevant here

[image]
[image]

haslinghuis, 2024-09-04:
- Unable to reproduce (firmware 4.6.0, configurator 11.0.0)
- What GPS unit is being used? It is using MSP ?
- Please provide support ID

SeriouSerg, 2024-09-04:
> * Unable to reproduce (firmware 4.6.0, configurator 11.0.0)
> * What GPS unit is being used? It is using MSP ?
> * Please provide support ID

The same firmware and the same configurator. 
GPS module UBLOX M10 (RUSHFPV GNSS MINI)connected to UART6

haslinghuis, 2024-09-04:
https://rushfpv.net/products/rushfpv-gnss-mini does not seem to use MSP.
That leaves a support ID (create on CLI tab and submit ID here).

haslinghuis, 2024-09-04:
```
GPS: connected, UART7 115200 (set to 115200), auto config OFF, version =  M10
```

- Not sure if it matters that auto config is disabled.
- Did you try latest master as there was an issue with MSP resolved today ?

SeriouSerg, 2024-09-04:
> ```
> GPS: connected, UART7 115200 (set to 115200), auto config OFF, version =  M10
> ```
> 
> * Not sure if it matters that auto config is disabled.
> * Did you try latest master as there was an issue with MSP resolved today ?

Yes, I cloned the repository a couple of hours ago

haslinghuis, 2024-09-04:
Ah, it's a local build. Try to replicate using cloud build.
Take care of the CPU temp. 81% is rather high and could lead to data integrity issues.

SeriouSerg, 2024-09-04:
> Ah, it's a local build. Try to replicate using cloud build. Take care of the CPU temp. 81% is rather high and could lead to data integrity issues.

I know, thanks. The temperature is about 49 deg in-flight.

ledvinap, 2024-09-04:
`serializeBoxNameFn` is one possible source of memory overwrite

also:
```
set led_inversion = 68
###ERROR IN dump: CORRUPTED CONFIG: led_inversion = 68###
```

SeriouSerg, 2024-09-05:
> `serializeBoxNameFn` is one possible source of memory overwrite
> 
> also:
> 
> ```
> set led_inversion = 68
> ###ERROR IN dump: CORRUPTED CONFIG: led_inversion = 68###
> ```

Yes, I’m also digging in this direction, thanks for the tip. And this variable (**led_inversion**) also constantly throws out the wrong value, this also seemed strange to me. I checked all the sources for writing to it, but found nothing.

ledvinap, 2024-09-05:
can you try capturing MSP boxnames response?

SeriouSerg, 2024-09-05:
> can you try capturing MSP boxnames response?

How to do it?

ledvinap, 2024-09-05:
Someone from @betaflight-configurator may help ... Or you can construct MSP request by hand and send it from serial terminal

haslinghuis, 2024-09-05:
https://github.com/betaflight/betaflight-configurator/blob/57d9a7e8c02ea834d5a65a43244c489536507ee0/src/js/msp/MSPHelper.js#L545-L560

SeriouSerg, 2024-09-05:
Friends, this is all good, but I do not know how to use the last two posts. Just so you understand, I got into the source code just a week ago to add a pitot tube. To look for errors, especially those where there is an overlap or memory leak - this will take forever with my current level of knowledge of how betaflight composed.

haslinghuis, 2024-09-05:
More test results (using MATEKH743):

> To Reproduce

    Enter `ALT_HOLD_MODE` in the Custom Defines area.
    Perform Cloud build.
    Put something into `craftName` parameter.
    Reboot and add GPS port to enable GPS.

- Checked with and without the dump (leaving out `led_inversion`)
- Why are LED resources added?
- `led_inversion` will be corrupted. `defaults group_id 505` or `defaults` does not recover corrupted config.
- In my case it does not affect `craftName`
- Also it happens without `ALT_HOLD_MODE` custom define.
- Build without [3505106](https://github.com/betaflight/betaflight/commit/350510603c85767041c2d3a3635dd339220da5a4) produces same result:

```
# version
# Betaflight / STM32H743 (SH74) 4.6.0 Sep  5 2024 / 16:25:42 (f0e1deb93) MSP API: 1.47
# config rev: 7366c14

# start the command batch
batch start

board_name MATEKH743
manufacturer_id MTKS

# name: MARK

# resources
resource LED 1 B00
resource LED 2 D13
resource LED 3 E05

# feature
feature GPS
feature LED_STRIP
feature OSD

# serial
serial 7 2 115200 115200 0 115200

# master
set led_inversion = 84
###ERROR IN diff: CORRUPTED CONFIG: led_inversion = 84###
set craft_name = MARK
```

ledvinap, 2024-09-05:
@haslinghuis : Does it break without configurator connection?

ledvinap, 2024-09-05:
[3505106](https://github.com/betaflight/betaflight/commit/350510603c85767041c2d3a3635dd339220da5a4) is unlikely to cause this

haslinghuis, 2024-09-05:
Did CLI only after fresh flash

```bash
set craft_name = MARK                 # clean diff
serial 7 2 115200 57600 0 115200      # clean diff
feature GPS                           # corrupted diff with added LED resources
```

SeriouSerg, 2024-09-05:
> is unlikely to cause this

Definitely no. I just checked.

> Did CLI only after fresh flash (save and reboot in between steps)
> 
> ```shell
> set craft_name = MARK                 # clean diff
> serial 7 2 115200 57600 0 115200      # clean diff
> feature GPS                           # corrupted diff with added LED resources
> ```

Yes. 

[image]

ledvinap, 2024-09-05:
From map file: 
```
                0x24008108                usbDetectPin.lto_priv.0
                0x240082a4                statusLedConfig_System
                0x240082bc                pilotConfig_System
 *fill*         0x240082de        0x2 
 .bss           0x240082e0      0xd5e /tmp/ccih0OrI.ltrans3.ltrans.o
                0x240082e0                pinioBoxRuntimeConfig.lto_priv.0
                0x240082e4                flightModeFlags
```

So maybe  something is overwriting this area of memory

ledvinap, 2024-09-05:
Can you try to replicate this without configurator? Using serial terminal (putty) to enter cli directly (send `#`  to enter cli mode)

SeriouSerg, 2024-09-05:
> ```shell
> feature GPS 
> ```

Done. There is no error.

[image]

SeriouSerg, 2024-09-05:
And then open configurator, go to cli and get error:

[image]

Return to terminal and here is error too. 

[image]

ledvinap, 2024-09-05:
easiest way now is to use debugger + write breakpoint on affected config.
but box name serialization is most likely (84 is ascii T)

SeriouSerg, 2024-09-05:
> easiest way now is to use debugger + write breakpoint on affected config. but box name serialization is most likely (84 is ascii T)

This is understandable, with a debugger, I would have caught this error yesterday if the firmware project had been in STM32CubeIDE. I have a debugger and I have a bunch of MCU and products based on them. But I work in CubeIDE and I know how to debug there, but I have no idea at all how to do it in Windows WSL (where I build firmware for betaflight)

If you tell me how to connect the debugger to WSL, then I'll take a look. I have the original ST-Link V3

SeriouSerg, 2024-09-05:
Just tried to flash firmware and config it without configurator (by terminal)

There is no errors:

[image]

So what's the verdict? Is there a problem with the firmware or a problem with the configurator?

ledvinap, 2024-09-05:
firmware, triggered by configurator.

SeriouSerg, 2024-09-05:
Fixed.

`msp_serial.h `
insufficient out buffer size. 
`#define MSP_PORT_OUTBUF_SIZE_MIN 320`
changed to 
`#define MSP_PORT_OUTBUF_SIZE_MIN 512`

[image]

PR #13879 (2024-09-10): Fix MSP output buffer overflow
- fixes #13875
closed 2024-09-10 by haslinghuis