betaflight/betaflight #13876: SITL version don't build

brainrom, 2024-09-05:
### Describe the bug

Undefined reference happens during building SITL target

```
$ make TARGET=SITL
make -j ./obj/betaflight_4.6.0_SITL.hex
make[1]: Entering directory '/home/kokokoshka/repos/betaflight'
EF HASH -> ./obj/main/SITL/.efhash_4d7648962c482093c07f7d5771c80f63
%% (optimised) lib/main/dyad/dyad.c 
%% (optimised) ./src/main/drivers/accgyro/accgyro_virtual.c 
%% (optimised) ./src/main/drivers/barometer/barometer_virtual.c 
%% (size optimised) ./src/main/drivers/serial_tcp.c 
%% (optimised) ./src/main/drivers/compass/compass_virtual.c 
%% (optimised) ./src/main/build/debug.c 
%% (optimised) ./src/main/build/build_config.c 
%% (optimised) ./src/main/build/debug_pin.c 
%% (optimised) ./src/main/build/version.c 
%% (optimised) ./src/main/target/SITL/sitl.c 
%% (optimised) ./src/main/target/SITL/udplink.c 
%% (size optimised) ./src/main/main.c 
%% (optimised) ./src/main/pg/adc.c 
%% (optimised) ./src/main/pg/alt_hold.c 
%% (optimised) ./src/main/pg/beeper.c 
%% (optimised) ./src/main/pg/beeper_dev.c 
%% (optimised) ./src/main/pg/board.c 
%% (optimised) ./src/main/pg/bus_i2c.c 
%% (optimised) ./src/main/pg/bus_quadspi.c 
%% (optimised) ./src/main/pg/bus_spi.c 
%% (optimised) ./src/main/pg/dashboard.c 
%% (optimised) ./src/main/pg/displayport_profiles.c 
%% (optimised) ./src/main/pg/dyn_notch.c 
%% (optimised) ./src/main/pg/flash.c 
%% (optimised) ./src/main/pg/gps.c 
%% (optimised) ./src/main/pg/gps_lap_timer.c 
%% (optimised) ./src/main/pg/gps_rescue.c 
%% (optimised) ./src/main/pg/gyrodev.c 
%% (optimised) ./src/main/pg/max7456.c 
%% (optimised) ./src/main/pg/mco.c 
%% (optimised) ./src/main/pg/motor.c 
%% (optimised) ./src/main/pg/msp.c 
%% (optimised) ./src/main/pg/pg.c 
%% (optimised) ./src/main/pg/piniobox.c 
%% (optimised) ./src/main/pg/pinio.c 
%% (optimised) ./src/main/pg/pin_pull_up_down.c 
%% (optimised) ./src/main/pg/rcdevice.c 
%% (optimised) ./src/main/pg/rpm_filter.c 
%% (optimised) ./src/main/pg/rx.c 
%% (optimised) ./src/main/pg/rx_pwm.c 
%% (optimised) ./src/main/pg/rx_spi.c 
%% (optimised) ./src/main/pg/rx_spi_cc2500.c 
%% (optimised) ./src/main/pg/rx_spi_expresslrs.c 
%% (optimised) ./src/main/pg/scheduler.c 
%% (optimised) ./src/main/pg/sdcard.c 
%% (optimised) ./src/main/pg/sdio.c 
%% (optimised) ./src/main/pg/serial_uart.c 
%% (optimised) ./src/main/pg/stats.c 
%% (optimised) ./src/main/pg/timerio.c 
%% (optimised) ./src/main/pg/timerup.c 
%% (optimised) ./src/main/pg/usb.c 
%% (optimised) ./src/main/pg/vcd.c 
%% (optimised) ./src/main/pg/vtx_io.c 
%% (optimised) ./src/main/pg/vtx_table.c 
%% (optimised) ./src/main/common/bitarray.c 
%% (optimised) ./src/main/common/colorconversion.c 
%% (optimised) ./src/main/common/crc.c 
%% (speed optimised) ./src/main/common/encoding.c 
%% (optimised) ./src/main/common/explog_approx.c 
%% (speed optimised) ./src/main/common/filter.c 
%% (optimised) ./src/main/common/gps_conversion.c 
[... 206 lines cut ...]
make[1]: Leaving directory '/home/kokokoshka/repos/betaflight'
make: *** [Makefile:579: hex] Error 2
```

SITL target don't implement IO methods.
In `cliSerialPassthrough`, `IOLo` and `IOHi` are used for timers poweroff, when enabling serial passthrough to ESC.
Commenting out this section, or implementing stub IOLo, IOHi is enough and after that, SITL operates normally. But I'm unsure what is the correct way to fix this bug.

### To Reproduce

Build SITL target `make TARGET=SITL`

### Expected behavior

SITL correctly builds and operates

### Support ID

```txt
Not applicable
```

### Flight controller

SITL

haslinghuis, 2024-09-05:
`make SITL` or make `TARGET=SITL` both produce an ELF.

haslinghuis, 2024-09-05:
```bash
make TARGET=SITL
make -j ./obj/betaflight_4.6.0_SITL.hex
make[1]: Entering directory '/home/mark/dev/betaflight/betaflight/fix-gpspassthrough/betaflight'
EF HASH -> ./obj/main/SITL/.efhash_4d7648962c482093c07f7d5771c80f63
%% (optimised) lib/main/dyad/dyad.c 
%% (optimised) ./src/main/drivers/accgyro/accgyro_virtual.c 
%% (optimised) ./src/main/drivers/barometer/barometer_virtual.c 
%% (optimised) ./src/main/drivers/compass/compass_virtual.c 
%% (size optimised) ./src/main/drivers/serial_tcp.c 
%% (optimised) ./src/main/build/build_config.c 
%% (optimised) ./src/main/build/debug.c 
%% (optimised) ./src/main/build/debug_pin.c 
%% (optimised) ./src/main/build/version.c 
%% (optimised) ./src/main/target/SITL/sitl.c 
%% (optimised) ./src/main/target/SITL/udplink.c 
%% (size optimised) ./src/main/main.c 
%% (optimised) ./src/main/pg/adc.c 
%% (optimised) ./src/main/pg/alt_hold.c 
%% (optimised) ./src/main/pg/beeper.c 
%% (optimised) ./src/main/pg/beeper_dev.c 
%% (optimised) ./src/main/pg/board.c 
%% (optimised) ./src/main/pg/bus_i2c.c 
%% (optimised) ./src/main/pg/bus_quadspi.c 
%% (optimised) ./src/main/pg/bus_spi.c 
%% (optimised) ./src/main/pg/dashboard.c 
%% (optimised) ./src/main/pg/displayport_profiles.c 
%% (optimised) ./src/main/pg/dyn_notch.c 
%% (optimised) ./src/main/pg/flash.c 
%% (optimised) ./src/main/pg/gps.c 
%% (optimised) ./src/main/pg/gps_lap_timer.c 
%% (optimised) ./src/main/pg/gps_rescue.c 
%% (optimised) ./src/main/pg/gyrodev.c 
%% (optimised) ./src/main/pg/max7456.c 
%% (optimised) ./src/main/pg/mco.c 
%% (optimised) ./src/main/pg/motor.c 
%% (optimised) ./src/main/pg/msp.c 
%% (optimised) ./src/main/pg/pg.c 
%% (optimised) ./src/main/pg/piniobox.c 
%% (optimised) ./src/main/pg/pinio.c 
%% (optimised) ./src/main/pg/pin_pull_up_down.c 
%% (optimised) ./src/main/pg/rcdevice.c 
%% (optimised) ./src/main/pg/rpm_filter.c 
%% (optimised) ./src/main/pg/rx.c 
%% (optimised) ./src/main/pg/rx_pwm.c 
%% (optimised) ./src/main/pg/rx_spi.c 
%% (optimised) ./src/main/pg/rx_spi_cc2500.c 
%% (optimised) ./src/main/pg/rx_spi_expresslrs.c 
%% (optimised) ./src/main/pg/scheduler.c 
%% (optimised) ./src/main/pg/sdcard.c 
%% (optimised) ./src/main/pg/sdio.c 
%% (optimised) ./src/main/pg/serial_uart.c 
%% (optimised) ./src/main/pg/stats.c 
%% (optimised) ./src/main/pg/timerio.c 
%% (optimised) ./src/main/pg/timerup.c 
%% (optimised) ./src/main/pg/usb.c 
%% (optimised) ./src/main/pg/vcd.c 
%% (optimised) ./src/main/pg/vtx_io.c 
%% (optimised) ./src/main/pg/vtx_table.c 
%% (optimised) ./src/main/common/bitarray.c 
%% (optimised) ./src/main/common/colorconversion.c 
%% (optimised) ./src/main/common/crc.c 
%% (speed optimised) ./src/main/common/encoding.c 
%% (optimised) ./src/main/common/explog_approx.c 
%% (speed optimised) ./src/main/common/filter.c 
%% (optimised) ./src/main/common/gps_conversion.c 
%% (optimised) ./src/main/common/huffman.c 
%% (optimised) ./src/main/common/huffman_table.c 
%% (speed optimised) ./src/main/common/maths.c 
%% (optimised) ./src/main/common/printf.c 
[... 175 lines cut ...]
%% (optimised) ./src/main/sensors/rangefinder.c 
%% (optimised) ./src/main/telemetry/telemetry.c 
%% (optimised) ./src/main/telemetry/frsky_hub.c 
%% (optimised) ./src/main/telemetry/hott.c 
%% (optimised) ./src/main/telemetry/jetiexbus.c 
%% (optimised) ./src/main/telemetry/smartport.c 
%% (optimised) ./src/main/telemetry/ltm.c 
%% (optimised) ./src/main/telemetry/mavlink.c 
%% (optimised) ./src/main/telemetry/msp_shared.c 
%% (optimised) ./src/main/telemetry/ibus.c 
%% (optimised) ./src/main/telemetry/ibus_shared.c 
%% (optimised) ./src/main/sensors/esc_sensor.c 
%% (size optimised) ./src/main/io/vtx.c 
%% (size optimised) ./src/main/io/vtx_rtc6705.c 
%% (size optimised) ./src/main/io/vtx_smartaudio.c 
%% (size optimised) ./src/main/io/vtx_tramp.c 
%% (size optimised) ./src/main/io/vtx_control.c 
%% (size optimised) ./src/main/io/vtx_msp.c 
%% (size optimised) ./src/main/cms/cms_menu_vtx_msp.c 
%% (size optimised) lib/main/google/olc/olc.c 
Linking SITL 
   text	   data	    bss	    dec	    hex	filename
 282333	  16471	  74240	 373044	  5b134	./obj/main/betaflight_SITL.elf
Creating HEX ./obj/betaflight_4.6.0_SITL.hex
```

brainrom, 2024-09-05:
Maybe GCC version or compiler flags are affected?

haslinghuis, 2024-09-05:
```
gcc --version 
gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
```

brainrom, 2024-09-05:
```
gcc --version 
gcc (GCC) 14.2.1 20240805
```

AJ92, 2024-10-13:
adding 
```
void IOHi(IO_t io){
  (void)io;
};
void IOLo(IO_t io){
  (void)io;
};
```
to sitl.c works around the undefined references. compiles, but i haven't run it.

closed 2024-10-22 by haslinghuis

zhouqingf, 2024-12-29:
$ make TARGET=SITL
make -j ./obj/betaflight_4.3.0_SITL_norevision.hex
make[1]: Entering directory '/cygdrive/d/222/betaflight'
%% (optimised) lib/main/dyad/dyad.c
%% (size optimised) ./src/main/drivers/accgyro/accgyro_fake.c
%% (size optimised) ./src/main/drivers/barometer/barometer_fake.c
%% (size optimised) ./src/main/drivers/compass/compass_fake.c
%% (size optimised) ./src/main/drivers/serial_tcp.c
%% (optimised) ./src/main/build/build_config.c
%% (optimised) ./src/main/build/debug.c
%% (optimised) ./src/main/build/debug_pin.c
%% (optimised) ./src/main/build/version.c
%% (optimised) ./src/main/target/SITL/target.c
%% (optimised) ./src/main/target/SITL/udplink.c
%% (size optimised) ./src/main/main.c
%% (optimised) ./src/main/pg/adc.c
%% (optimised) ./src/main/pg/beeper.c
%% (optimised) ./src/main/pg/beeper_dev.c
%% (optimised) ./src/main/pg/board.c
%% (optimised) ./src/main/pg/bus_i2c.c
%% (optimised) ./src/main/pg/bus_quadspi.c
%% (optimised) ./src/main/pg/bus_spi.c
%% (optimised) ./src/main/pg/dashboard.c
%% (optimised) ./src/main/pg/displayport_profiles.c
%% (optimised) ./src/main/pg/dyn_notch.c
%% (optimised) ./src/main/pg/flash.c
%% (optimised) ./src/main/pg/gyrodev.c
%% (optimised) ./src/main/pg/max7456.c
%% (optimised) ./src/main/pg/mco.c
%% (optimised) ./src/main/pg/motor.c
%% (optimised) ./src/main/pg/pg.c
%% (optimised) ./src/main/pg/pin_pull_up_down.c
%% (optimised) ./src/main/pg/pinio.c
%% (optimised) ./src/main/pg/piniobox.c
%% (optimised) ./src/main/pg/rcdevice.c
%% (optimised) ./src/main/pg/rx.c
%% (optimised) ./src/main/pg/rx_pwm.c
%% (optimised) ./src/main/pg/rx_spi.c
%% (optimised) ./src/main/pg/rx_spi_cc2500.c
%% (optimised) ./src/main/pg/rx_spi_expresslrs.c
%% (optimised) ./src/main/pg/scheduler.c
%% (optimised) ./src/main/pg/sdcard.c
%% (optimised) ./src/main/pg/sdio.c
%% (optimised) ./src/main/pg/serial_uart.c
%% (optimised) ./src/main/pg/stats.c
%% (optimised) ./src/main/pg/timerio.c
%% (optimised) ./src/main/pg/timerup.c
%% (optimised) ./src/main/pg/usb.c
%% (optimised) ./src/main/pg/vcd.c
%% (optimised) ./src/main/pg/vtx_io.c
%% (optimised) ./src/main/pg/vtx_table.c
%% (optimised) ./src/main/common/bitarray.c
%% (optimised) ./src/main/common/colorconversion.c
%% (optimised) ./src/main/common/crc.c
%% (speed optimised) ./src/main/common/encoding.c
%% (optimised) ./src/main/common/explog_approx.c
%% (speed optimised) ./src/main/common/filter.c
%% (optimised) ./src/main/common/gps_conversion.c
%% (optimised) ./src/main/common/huffman.c
%% (optimised) ./src/main/common/huffman_table.c
%% (speed optimised) ./src/main/common/maths.c
%% (optimised) ./src/main/common/printf.c
%% (optimised) ./src/main/common/printf_serial.c
%% (speed optimised) ./src/main/common/sdft.c
%% (optimised) ./src/main/common/sensor_alignment.c
%% (optimised) ./src/main/common/streambuf.c
%% (optimised) ./src/main/common/string_light.c
%% (optimised) ./src/main/common/strtol.c
%% (optimised) ./src/main/common/time.c
%% (speed optimised) ./src/main/common/typeconversion.c
[... 203 lines cut ...]
%% (optimised) ./src/main/telemetry/ltm.c
%% (optimised) ./src/main/telemetry/mavlink.c
%% (optimised) ./src/main/telemetry/msp_shared.c
%% (optimised) ./src/main/telemetry/ibus.c
%% (optimised) ./src/main/telemetry/ibus_shared.c
%% (optimised) ./src/main/sensors/esc_sensor.c
%% (size optimised) ./src/main/io/vtx.c
%% (size optimised) ./src/main/io/vtx_rtc6705.c
%% (size optimised) ./src/main/io/vtx_smartaudio.c
%% (size optimised) ./src/main/io/vtx_tramp.c
%% (size optimised) ./src/main/io/vtx_control.c
%% (size optimised) lib/main/google/olc/olc.c
Linking SITL
lto-wrapper: warning: using serial compilation of 5 LTRANS jobs
lto-wrapper: note: see the ‘-flto’ option documentation for more information
   text    data     bss     dec     hex filename
 326085    4036   75680  405801   63129 ./obj/main/betaflight_SITL.elf
Creating HEX ./obj/betaflight_4.3.0_SITL_norevision.hex
objcopy: obj/betaflight_4.3.0_SITL_norevision.hex 64-bit address 0x100401000 out of range for Intel Hex file
objcopy: obj/betaflight_4.3.0_SITL_norevision.hex: bad value
make[1]: *** [Makefile:354: obj/betaflight_4.3.0_SITL_norevision.hex] Error 1
make[1]: Leaving directory '/cygdrive/d/222/betaflight'
make: *** [Makefile:575: hex] Error 2

您好，我在编译的过程中遇到了该问题

PR #13972 (2024-10-22): Fix SITL build
- fixes #13876 and https://github.com/betaflight/betaflight-configurator/issues/4215
- documentation https://github.com/betaflight/betaflight.com/pull/475
PR #13992 (2024-10-22): [4.5.2] Fix SITL build
- backport of  #13972
- fixes #13876 and https://github.com/betaflight/betaflight-configurator/issues/4215
- documentation https://github.com/betaflight/betaflight.com/pull/475