betaflight/betaflight #13888: AT32F435G Flight Nazgul ECO Ibata Offset = -19000

Fpv191, 2024-09-09:
### Describe the bug

iFlight Nazgul ECO Ibata Offset = -19000

This is set from the factory on every one as far as I can determine with other samples or comments.

( Although iFlight support deny this -19000 setting)

I would normally expect a value of Ibata Offset = 0  or a much smaller value than -19000 (-19A)

If set to Ibata Offset = 0  results in 21A at idle or disarmed motors.

### To Reproduce

Set to -19000 results 2 A at idle, which is also inaccurate.

[image]
[image]
[image]
[image]
[image]

https://1drv.ms/v/s!Au08QAdcoH-UkG0JH7ZTDmYmnlyb

https://1drv.ms/v/s!Au08QAdcoH-UkG6DZyWXeYowBmef

### Expected behavior

To see under 1A disarmed. 
(Accounting only for VTX and cam current draw etc.)

### Support ID

```txt
Id: "1b2cbdb1-16af-46ae-9a46-10cfa4b9107d"
```

### Flight controller

AT32F435G iFlight Blitz stack E55S 4-IN-1 ESC

### Other components

iFlight ELRS 915 Rx
Blitz 1.6w Analog VTX

### How are the different components wired up (including port information)

All standard wiring as delivered out of the box from factory using plugs for Nazgul ECO .

### Add any other context about the problem that you think might be relevant here

Is this a Betaflight bug with this version?
 AT32F435 4.5.0 Aug 10 2023 / 07:39:06 (50da879b1c)
IFLIGHT_BLITZ_F435

I'm not looking to update yet and potentially lose this factory version which I cannot find as a backup hex.

Has this issue been addressed in official 4.5.0 releases?

Hopefully you can replicate this bug with this iFlight Blitz AT435 stack hardware and address this in future builds if needed.

Thanks guys for all you do for this hobby.
Cheers.

Fpv191, 2024-09-09:
AT32F435 4.5.0 Aug 10 2023 / 07:39:06 (50da879b1c)
IFLIGHT_BLITZ_F435

ledvinap, 2024-09-09:
IMO this is expected behavior on boards with bidirectional current measurement IC.
It can be improved on HW or FW level [differential ADC input with ratiometric reference / use VCC (if used to power current sensor ic) as analog reference for current measurement], but it should provide usable data without changes. Just calibrate offset.

Fpv191, 2024-09-09:
Thanks for your fast reply. I only brought this to your attention out of concern that I have never needed this ibata offset value other than 0 in over 10 years of building or flying quads. If you determine this is acceptable, then no problems. Thanks for confirming.

Helmars, 2024-09-15:
Hello,

I have the same issue, but I noticed a pattern. I have 4s battery and it shows 13A at 15.2V and then slowly drops as the battery discharges. 15.2/13 is the same ratio as 24.4/20.9 for issue reporter. It looks like current is incorrectly calculated using **voltage** ADC input.

Helmars

closed 2024-09-22 by Fpv191

mixer3d, 2024-12-16:
Hi, just to let you know i saw recently similar issue with the same stack from iflight chimera 9 eco. But that case by default you can see 10.4A while quad is taking 0.3A, value Issue it's the same with FW 4.5.0 and 4.5.1 available for that FC. The offset to set is also more than 10k, don't remember at the moment. I contacted support and they told me, that it's "normal" and the FC/ESC are calibrated in a way, that real current usage is visible only while flying. So while hovering one can read some 16-18A, stil with throttle 0% it's 10.4A. Recommended calibration way with 5 steps did nothing besides or changing a little range.

mixer3d, 2024-12-16:
Voltage reading by default is more or less correct +/- 0.3V/6S