betaflight/betaflight #13901: Octocopter OCTO Flat + motor directions/yaw handling reversed

Patronics, 2024-09-14:
### Describe the bug

For octocopters with the `OCTO Flat +` configuration, the motors must spin in the opposite direction of what is shown in the diagram, in order for stable flight to be possible. With the motor directions configured as indicated, the yaw is reversed, leading to immediate crashes with out of control spinning.

### To Reproduce

Configure an octocopter with the Octo Flat + configuration, with motors spinning in the directions indicated in the configurator, and attempt to fly it, it will start to spin and rapidly crash (or, potentially safer, hold the drone and observe that the yaw direction it twists when given yaw controls from the remote)

### Expected behavior

The drone is able to fly and is able to control its yaw orientation accurately, not spinning out of control or to the reverse of the given inputs.

### Support ID

```txt
22f3e59e-8aa2-476a-86df-652b9d8811fd
```

### Flight controller

SpeedyBee F405 V4

### Other components

Two Speedybee 50A 4-in one ESCs, running the latest release BlueJay version. Motors and frame are from the MikroKopter Oktokopter XL.

### How are the different components wired up (including port information)

Motor outputs 1-4 go to one of the 4-in-1 ESCs, and outputs 5-8 go to the other.

### Add any other context about the problem that you think might be relevant here

This is with the latest Betaflight firmware release, version 4.5.1. I saw that the release 4.5.0 added a new layout for octocopters (Octo X8 +), I wonder if that release accidentally changed something with the existing configuration? I haven't tested on any other firmware versions. 

As a workaround once I realized the issue, I was able to toggle the "motor direction is reversed" switch in the motors tab, however this also reversed the direction the motors are shown to spin in the diagram, so it no longer reflects reality. The motors still spin in the same direction they did originally, but with that toggle active, the firmware handles the Yaw direction properly. 

Here's a screenshot of the blackbox recording of the issue, showing that the motors increasingly were driven in order to attempt to counteract the yaw, but instead were making it much worse until crashing.
[image]

haslinghuis, 2024-09-15:
@VirtualEnder any advice on this one?

nerdCopter, 2024-09-16:
Octo 8+ code did not changed the flat: https://github.com/betaflight/betaflight/pull/12175/files

VirtualEnder, 2024-09-16:
> Octo 8+ code did not changed the flat: https://github.com/betaflight/betaflight/pull/12175/files
> 
> EDIT: may need to verify configurator is not setting the wrong mode, or maybe just reverse ESC firmware for said motors.

Yes can confirm, no code changes to the octo + flat code, and there don't seem to be any errors in the code that would cause any improper assignments of the layout that I can find. Possibly a motor rotation direction or motor order assignment problem, as @nerdCopter suggests. Please confirm all of those and if it is still not operating as expected check back.

haslinghuis, 2024-09-16:
Yes sorry for pinging - see now it's octo flatp instead of octo x8p

Patronics, 2024-09-17:
> Possibly a motor rotation direction or motor order assignment problem, as @nerdCopter suggests. Please confirm all of those and if it is still not operating as expected check back.

If you are referring to me when you ask to confirm those, I definitely confirmed several times that the physical motor order did match the order in the diagram. I don't see how the ESC's configuration is a factor, since the motors all spin in the direction that the propellers are installed in and do achieve lift. The only issue is the handling of them in terms of Yaw in Betaflight, where it only behaves as expected when it's configured to believe that the motors are spinning in the opposite direction of how they really are. 

Just to confirm as a sanity check, the view in the Betaflight configuration of the motor configuration and spin direction is meant to be as viewed from above, right? Like looking down at the drone sitting on the table? 

If it's meant to be viewed from below, like the drone was flying overhead, then the rotation view shown would match reality (but that would also mean that the configured motor positions are mirrored from reality). I've watched some videos where they seemed to work with the same assumption I did, that it was being viewed from above, but I think with the symmetry of most quadcopters it's a bit less of an issue (at least regarding spin direction) than with the octocopter configuration. If I was wrong with that assumption, it'd probably be good to have that clearly documented somewhere, since it's a very easy mistake to make.

nerdCopter, 2024-09-17:
did 4.4.x work? 4.3.x?  this would help with investigation.

Patronics, 2024-09-17:
> did 4.4.x work? 4.3.x? this would help with investigation.

I haven't used this drone with any versions before the current 4.5.1, so I'm not sure. I might be able to test installing some older versions at some point, but given the high risk of crashes if the configuration is still reversed, I'm hesitant to do so.

Quick-Flash, 2024-09-17:
> I haven't used this drone with any versions before the current 4.5.1, so I'm not sure. I might be able to test installing some older versions at some point, but given the high risk of crashes if the configuration is still reversed, I'm hesitant to do so.

You should be able to arm with the props off and see if it responds as it should in the yaw direction. No risk of crashes.

CBird94, 2024-10-21:
> did 4.4.x work? 4.3.x? this would help with investigation.

@nerdCopter I just tried this on 4.5.1, 4.4.3, and 4.3.2 with my OCTO FLAT + drone after experiencing the same issue as @Patronics with the drone immediately losing control of yaw on takeoff. These betaflight versions all seem to be impacted with yaw being reversed. 

I'm using a speedybee f7 v3 FC, 2x TBS lucid AM32 ESCs. I confirmed that I'm using the correct motor direction and order that betaflight expects. 

Attached are 4 logs: 1 real flight, and 3 arms on the bench without props per Quick_Flash's advice. 
[OctoFlat+Yaw_Issue.zip](https://github.com/user-attachments/files/17465585/OctoFlat%2BYaw_Issue.zip)

Patronics, 2024-10-21:
@CBird94 thanks for confirming. 

As an update on my drone's behavior, I realized the Octo flat + isn't the correct mixer for my octocopter, as the rotors aren't symmetrically spaced from the center, but instead more of a square arrangement.

I didn't encounter the same issue with the custom mixer, which likewise resolved the other issues I was having due to the wrong layout. (I was a bit worried that the yaw issue I'd observed was also caused by the incorrect mixer, though I didn't think it would make that significant a difference).

Patronics, 2024-10-21:
In case it's relevant, from a quick search through the code I found this logic. https://github.com/betaflight/betaflight/blob/69f75b55f651fa9cd2925f08a5b484ef97c53e56/src/main/flight/mixer_init.c#L181

This mixer array seems to be in a different order than the entries are listed in the configurator, though I'm not sure if there's any additional logic that rearranges this array to the correct order. In the configurator (and in a custom mixer for the same motor layout) the motor numbered 1 is the left motor, followed by the front motor, then the right one, then the rear, while this configuration array lists the diagonal motors first (and then lists the front motor first as motor 5, then right, then rear, then left) , so the so perhaps that's the issue? Perhaps someone with more familiarity with the codebase can confirm if I'm on the right track in tracing down the cause of this issue, or if there's some other logic that handles reordering the array entries?

haslinghuis, 2024-10-21:
Entries in configurator are sorted.

Patronics, 2024-11-22:
Since I've determined that my drone needs the custom layout for it's motors, it's somewhat impractical for me to do more testing, but I would be happy to remove the propellers and do some testing when I get a chance if that would still be useful, let me know what I should test if so.

Douglas-House, 2024-12-26:
I came across this issue after encountering a very similar problem with my new custom multirotor using the OCTOFLATP mixer. Taking off the propellers and testing on the ground, I can confirm that the default yaw direction is inverted, Eg. when a sudden CW yaw direction is requested from the stick, a sudden jitter in the CCW direction is noted on the drone body (in addition to general wobbliness/instability). I can also confirm that ticking the "motor direction is reversed" switch in the betaflight configurator motors tab without adjusting anything else fixes this issue, preventing throttle runaway.

Patronics, 2025-01-26:
Unless anything has been changed to resolve this issue, I don't think there's any reason to close it. Is there any way to suppress the bot from repeatedly trying to close it, which only hides the problem without fixing anything?