betaflight/betaflight #13919: Yellow GPS icon is not displayed despite GPS fix

DaDuuuke, 2024-09-22:
### Describe the bug

GPS outputs correct data, but is not listed as a participant in the sensor bar. This also means that no data is displayed in the OSD even though the FC processes it. It is the DarwinFPV FoldApe
[image]

### To Reproduce

UBLOX and NMEA were tried. Data is only output via UBLOX. Baud rates of 115200 and 57600 were also tried, unfortunately both were fixed but the icon or display in the OSD was not displayed.

### Expected behavior

Just like I know it from my other FCs of the 411 and it works there too.

### Support ID

```txt
ID: "1e088b9b-66a5-456a-969a-dd20217810eb"
```

### Flight controller

Darwinfpv F411

### Other components

DarwinFPV FoldApe4 ELRS (With DJI O3 Air Unit - build in by myself)
GPS: Darwinfpv M8N GPS
RX: Onboard SPI ELRS 2.4G (Version 3.X)

### How are the different components wired up (including port information)

[image]

haslinghuis, 2024-09-22:
- From status command in cli:
```
GPS: NOT CONNECTED, UART2 38400 (set to 115200), NOT CONFIGURED, version =  unknown
Arming disable flags: RXLOSS CLI MSP
```

- Does it work on either 38400 or 115200 and auto-config disabled ?
- Changing GPS configuration / baud rate require a cold reboot for changes to take effect.

DaDuuuke, 2024-09-22:
It works only on 115200 with no auto config. I have red the Darwin manual for the M8N and did everything as it is mentioned there. RX from GPS to TX2 on FC an TX from GPS to RX2 on FC (wiring checked). 
I had GPS position many times through the window but the GPS icon not even was red (as you can see in the screen).

DaDuuuke, 2024-09-23:
> * From status command in cli:
>
> ```
> GPS: NOT CONNECTED, UART2 38400 (set to 115200), NOT CONFIGURED, version =  unknown
> Arming disable flags: RXLOSS CLI MSP
> ```
>
> * Does it work on either 38400 or [...]

I tried all combinations, even a sensor readout via u-centersetup for M8N GPS Modules but Betaflight wont show/recognize the sensor as you can see in my 2 screens. first screen is the "status" readout, second screen shows the correct GPS position at the same time?! I dont get it :( 
[image]
[image]

ctzsnooze, 2025-05-03:
Try a different gps module.