betaflight/betaflight #13921: MAMBA F405 MINI MK2 i2C (SDA, SCL) enabling

PeterPutoAmo, 2024-09-23:
### Describe the bug

Hi

I'm trying to enabling GPS rescue function on a MAMBA F405 **MINI** MK2 - based drone. To do so, having a barometer is needed for accurate altitude control. All barometers use i2C, mine, a BMP180 too. 

Searching for the pinOut of this FC no SDA o SDL mins are listed, however the board has a "i2C" label, two pads going to pins (58) PB6 & (59) PB7 of microcontroller + pull up restistors. Those are listed as programmable SDL SDA pins on the microcontroller.

[image]
[image]

### To Reproduce

I connected a BMP180 to these 
GND -- GND
VCC -- 3'3V (this unit is 1'8-3'6v)
SCL -- PB6
SDA -- PB7

Enabled Barometer on Beta flight. No baro sensor data.

Checking wt a FLUKE multimeter no clock is found on SCL pin. Baro is receiving power.

### Expected behavior

Barometer working

### Support ID

```txt
#
```

### Flight controller

MAMBA F405 MINI MK2

haslinghuis, 2024-09-23:
- Please provide support ID as requested when opening the issue.
- Need to add #define `USE_BARO_BMP388` and also `USE_BARO` if the config does not have baro defined..
- PB8/PB9 also provide SCL/SDA for I2C 1

PeterPutoAmo, 2024-09-23:
Hi Mark, thaks for the quick reply.

- I never asked before on GitHub. I don´t know what that ID is.
- I think i did it now correctly, following Joshua [video ](https://www.youtube.com/watch?v=KRMeEbgq-gc&ab_channel=JoshuaBardwellLivestreamClips)
- Ok

Baro BMP180 is not woking yet. No i2C device found

[image]
I also tried after flashing 
[image]

No luck

haslinghuis, 2024-09-23:
Try 4.3.2 or core build to expose all drivers to be sure.

PeterPutoAmo, 2024-09-25:
Hi. 

I installed 4.3.2 and baro is working now. Thanks!

Do you need any more info for fixing the lastest version?

haslinghuis, 2024-09-25:
Post support ID from this one.

closed 2024-11-02 by github-actions[bot]