betaflight/betaflight #13927: Change GPS to GNSS

jonathanoeijoeng, 2024-09-24:
### Is your feature request related to a problem? Please describe

It's just wording but I think GNSS is more correct then GPS since it can use other satellite from Rusia and China.

### Describe the solution you'd like

Change word GPS to GNSS

### Describe alternatives you've considered

Change word GPS to GNSS

haslinghuis, 2024-09-24:
[image]

jonathanoeijoeng, 2024-09-24:
Thanks for the table. So I means GNSS is more correct since it use multiple operatorrs comparing to GPS that only use 1 operator.

ctzsnooze, 2025-05-03:
Problem is that nearly every pilot in the world knows what GPS means, almost none have heard of GNSS.

jonathanoeijoeng, 2025-05-03:
But definitely it can be changed.  DJI already started to use GNSS instead
of GPS.

closed 2026-09-26 by blckmn