betaflight/betaflight #13962: Reorder Motors Wizard Not Working, but motors are working in test mode

shahmirthesquid, 2024-10-08:
### Describe the bug

Hello I am on latest betaflight 4.5.1 on OMNIBUSF4SD clone. This issue occured on the stock 4.5.0 firmware that came with this board. When I go into motor test mode, everything is working. But when I try to reorder motors to get them into the right directions, the motors just dont spin. I've tried adjusting the min throttle values but that doesn't work. My ESC is a BLH02102 which is a Fairycolor 30A clone. It runs on OneShot125.

### To Reproduce

1. Click "Reorder Motors"
2. Motors dont spin :'(
3. Try motor test on the side
4. all motors work??

### Expected behavior

Expected behaviour is that after I click reorder motors, one of the motors should start moving so i can recognize it on the screen.

### Support ID

```txt
Id: "a15e3193-0f95-43c3-8871-e89a56cdb5da"
```

### Flight controller

OMNIBUSF4SD-ICM

haslinghuis, 2024-10-08:
> It runs on OneShot125.

Reorder motor wizard only works with Dshot protocol commands.

closed 2024-10-08 by haslinghuis

shahmirthesquid, 2024-10-08:
No that's not true. I was able to resolve this issue and get motor wizard working again for me. 

First I went into my BLheli config and made the PPMs wider
[image]

Then I did the OneShot calibration according to docs: https://betaflight.com/docs/development/oneshot
[image]

This fixed all my problems. Now motor wizard works, and my motors feel better than before. As well as moving my min throttle value from 1140 down to something more normal at 1009. 

Hope this helps someone out there

haslinghuis, 2024-10-08:
You may be right as I could not find a trace in reorder motors wizard for dshot as it's motor direction that requires dshot protocol.