betaflight/betaflight #13984: battery warning/min battery  - over time

popbangshwang, 2024-10-19:
### Is your feature request related to a problem? Please describe

Greetings, I fly mostly micros and setting up the battery warnings can be a challenge at times. A punch out over the house or similar might sag the battery just enough when I'm further into my pack, resulting in a brief bleet of battery warning. Another is setting that value low enough as to not trigger all the time, but allow you enough time to get the drone in within the comfortable battery voltage.

### Describe the solution you'd like

Simply would it be possible to introduce a battery warning as it currently stands - lets say I set it to 3.3v and then an additional over time option, say I set it to 3 seconds. That allows me to enjoy the punch out later into my flight, while not having the warning pop up or beep. This still allows me to set sane battery limits while so that I can get it in safely, while still enjoying the pack without warnings every punch out.

### Describe alternatives you've considered

I think the alternative is what we currently have - You set the limits low enough, so that it doesn't trigger most of the time, but later into the flight you just expect to feel the warnings. The result is you fly to 3.3v or whatever on your 1s pack, which gives you just enough time to get the drone in before that 1s pack drops off a cliff.

Quick-Flash, 2024-10-30:
Already a feature

haslinghuis, 2024-10-30:
Checkout `vbat_sag_compensation`

https://betaflight.com/docs/wiki/tuning/4-2-Tuning-Notes#dynamic-battery-sag-compensation

Quick-Flash, 2024-10-30:
> Checkout `vbat_sag_compensation`
> 
> https://betaflight.com/docs/wiki/tuning/4-2-Tuning-Notes#dynamic-battery-sag-compensation

Wrong feature `vbat_duration_for_warning` and `vbat_duration_for_critical` is the feature that means you have to be in a critical or warning battery voltage for x amount of time before it will say you are in a critical or warning voltage.

```
if (cmp32(millis(), lastVoltageChangeMs) >= batteryConfig()->vbatDurationForWarning * 100) {
                    voltageState = BATTERY_WARNING;
                }
```
Based on this code I think that the units are 1/10th of a second, so that 10 is 1 second and 100 is 10 seconds.