betaflight/betaflight #13985: Motors RPM going up and down non-stop when arming at zero throttle

devgioele, 2024-10-19:
### Describe the bug

Inside the betaflight configurator, the setpoint received from the transmitter is correct and the motors spin as expected when the sliders are moved.
However, when disconnected from the configurator, as soon as I toggle the arming switch, the motors spin like crazy. The RPMs continue to go up and down. This is leaving all sticks at their default position and leaving throttle at zero. If I do not disarm the drone, the motor start overheating.

This happens without any airmode.
Changing the motor idle percentage between 0% and 10% does not solve the issue. 

Looking at the log file, you can see that there is some pattern in how the motors spin, but I have no idea why they spin in such a way.

### To Reproduce

Apply the same configuration on a F405.
Connect the battery.
Arm the drone.
Observe the motor spinning to the point of overheating.

### Expected behavior

On arming the drone, the a leveled surface, the motor should spin at a fairly constant speed or not spin at all.

### Support ID

```txt
f0001ec6-2a02-4dfa-a21d-0f07822c59d7
```

### Flight controller

SpeedyBee F405 clone bought from Aliexpress: https://it.aliexpress.com/item/1005006440609551.html

### Other components

ESC firmware: BLHELI_S
ESC protocol: DSHOT 300 (DSHOT 600 did not change anything)
Battery: 6S LiPo

### Add any other context about the problem that you think might be relevant here

The following issue might be related #11655

devgioele, 2024-10-19:
[log.zip](https://github.com/user-attachments/files/17444754/log.zip)

peterlozano, 2024-10-29:
I had this problem once and the cause was that the capacitor was not well soldered and one of the wires had detached from the FC pads.

devgioele, 2024-11-02:
Thank you for you answer. The motors are piloted correctly when using the sliders of betaflight configurator. Therefore I do not think that it is a hardware problem

nerdCopter, 2024-11-02:
- turn down D-slider to 0.7 and tune the quad.
  - https://hackmd.io/@nerdCopter/rJzic_N-p
- might be over-filtering causing feedback loop.
  - 3 dynamic gyro notches and 3 rpm harmonics are too many. set 1 dynamic gyro notch.
- massive D-term noise at 80Hz roll and 95Hz pitch.
- i believe this to be a tuning or hardware problem, not a software bug and therefore should be discussed on betaflight discord (https://discord.betaflight.com/), not github.

closed 2024-11-02 by nerdCopter