betaflight/betaflight #14014: HDZ AIO5 hangs or locks up while accessing the OSD BF4.6

Justicefpv, 2024-11-09:
### Describe the bug

The HDZ AIO5 flight controllers consistently freeze when modifying OSD settings in BetaFlight. I've confirmed this issue on two boards using the latest master build. While not occurring every time, the lockup happens frequently enough to be problematic - simply entering the OSD menu and changing any setting can freeze the FC, requiring a power cycle to recover.

### To Reproduce

To reproduce the issue: Enter the OSD menu and change any setting. The problem occurs more frequently if you first fly for a minute, land, then enter the OSD and scroll through settings.

### Expected behavior

The FC should not lockup

### Support ID

```txt
Id: "79ed5adf-1308-48bf-8580-69ae422ac56f"
```

### Flight controller

HDZ AIO5 (CRAZYBEEF4SX1280)

fobius006, 2024-11-15:
I’ve experienced this too, and when it happens, the VTX control through the VTX table also doesn’t work.

nerdCopter, 2024-12-03:
cross-referencing the other HDZ AIO issue: https://github.com/betaflight/betaflight/issues/14025

and adding that this youtuber reports settings revert upon reboot: https://youtu.be/WhrRTDBVz3A?feature=shared&t=303

wondering if ELRS backpack is enabled or not.

haslinghuis, 2025-01-03:
Fixed in #14091

closed 2025-01-03 by haslinghuis