betaflight/betaflight #14017: OSD Freezes on Betaflight logo on latest dev

DangerD1024, 2024-11-11:
### Describe the bug

OSD Freezes on Betaflight logo on latest dev.

To make OSD work i just reboot powered drone from Betaflight configurator or using rc menu (i actualy don't see it, i just remember menu item)

### To Reproduce

Just power on drone even with default settings

### Expected behavior

betaflight logo disappears and works properly

### Support ID

```txt
###

# # Problem description

# # OSD Stucks

# ###

# version
# Betaflight / STM32F7X2 (S7X2) 4.6.0 Nov 11 2024 / 13:14:12 (778f93f11) MSP API: 1.47
# config rev: 6239460
# board: manufacturer_id: DIAT, board_name: MAMBAF722_2022B

# status
MCU F722 Clock=216MHz, Vref=3.30V, Core temp=44degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3868, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=ICM42688P, ACC=ICM42688P, BARO=DPS310
OSD: MSP (53 x 20)
System Uptime: 15 seconds, Current Time: 2024-11-11T12:00:47.605+00:00
CPU:31%, cycle time: 124, GYRO rate: 8064, RX rate: 0, System rate: 10
Voltage: 2248 * 0.01V (6S battery - OK)
I2C Errors: 0
FLASH: JEDEC ID=0x00ef4018 16M
Arming disable flags: RXLOSS CLI MSP

# flash_info
Flash sectors=256, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=16777216 JEDEC ID=0x00ef4018
Partitions:
  0: FLASHFS   0 255
FlashFS size=16777216, usedSize=0

# dump master

# version
# Betaflight / STM32F7X2 (S7X2) 4.6.0 Nov 11 2024 / 13:14:12 (778f93f11) MSP API: 1.47
# config rev: 6239460

# start the command batch
batch start

board_name MAMBAF722_2022B
manufacturer_id DIAT

# name: -

# resources
resource BEEPER 1 B02
resource MOTOR 1 C08
resource MOTOR 2 C09
resource MOTOR 3 A08
[... 945 lines cut ...]
Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 3
    CH4 : DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

#
```

### Flight controller

MAMBAF722_2022B

DangerD1024, 2024-11-11:
> Please provide support ID as requested while opening bug report.

done

haslinghuis, 2024-11-11:
> Betaflight / STM32F7X2 (S7X2) 4.6.0 Nov  5 2024 / 18:22:01 (norevision) MSP API: 1.47

Support ID does not expose build key. Local builds are not supported.

closed 2024-12-20 by github-actions[bot]