betaflight/betaflight #14025: MSP VTX control via VTX table, BF 4.6 HDZ Support ID  If missing any drivers for GYRO, ACC, MAG, BARO, FLASH or OSD it's required to flash a Core Only build for your board.  Go to CLI tab and use the Submit Support Data button to get an Support ID. Please post the Support ID here.  Don't add three backticks or any other format symbol at the start or end of the block, they will be added automatically.ero AIO5

fobius006, 2024-11-15:
### Describe the bug

I’m using a VTX table to control power (3-position switch: 0/25/200), and in my mixes, the lower position of the switch outputs a value of 1500 on the channel, which sets the VTX power to 0 for that value. This is done so that when you power the drone from the battery, the video transmitter is always at 0 power, even when the radio isn’t connected. 

On 4.6, when I start the drone with VTX power at 0, it doesn’t switch power through the switch afterward. However, if I connect the drone with the switch already set to 25mW, everything works. Sometimes, when switching power to 0 with the switch, it doesn’t return to 25mW afterward.

### To Reproduce

Start drone with selected VTX 0 power 

### Expected behavior

Change power independents of starting VTX power 

### Support ID

```txt
f892a414-9abe-473f-b243-eb8985e59ede
```

### Flight controller

HDZero AIO5

nerdCopter, 2024-11-15:
please edit the description to include HDZero firmware version is flashed.

fobius006, 2024-11-15:
Good day, it`s last one for AIO5 1.7.0

haslinghuis, 2025-01-03:
Fixed in #14091

closed 2025-01-03 by haslinghuis