betaflight/betaflight #14033: Consider re-introducing 6 rates profiles

teneketo, 2024-11-18:
### Is your feature request related to a problem? Please describe

Exact same "issues" as https://github.com/betaflight/betaflight/issues/12316
Alot of (now apparently outdated) information out there suggests this should be possible. There are videos and articles about using a 6pos switch to change the 6 profiles. Took me week+ to randomly stumble upon the above link, suggesting this is no longer possible

### Describe the solution you'd like

Please consider re-introducing 6 rate profiles again. More and more radios have 6 pos switches now. For my personal use case I'd prefer to have the flexibility when flying cinematic out in the field: slow and steady -> more agile / with camera -> without camera / Everything 100%

### Describe alternatives you've considered

4 is good, 6 is better

### Other information

Thank you, I understand "No" is a viable answer too <3

anardine, 2025-02-03:
@teneketo I'm thinking about suitable use cases for all six... Do you envision this needed as something to gradually change? So you could go gradual between cine to aggressive?

nerdCopter, 2025-02-04:
while it is easy to modify firmware code to allow for custom count via cloud-build with something like
```
#ifndef CONTROL_RATE_PROFILE_COUNT
#define CONTROL_RATE_PROFILE_COUNT  4
#endif
```
near `src/main/target/common_pre.h:304` and `src/test/unit/platform.h:46`, upon my attempts, i noticed that the `CONTROL_RATE_PROFILE_COUNT` is not written to MSP and thus Configurator will not comply with variable number of rate profiles as it sits.

haslinghuis, 2025-02-04:
> Backward compatibility to existing 4.6-dev builds might also be required.

Users on 4.6-dev should always use master with rolling release of configurator.

nerdCopter, 2025-02-04:
> Users on 4.6-dev should always use master with rolling release of configurator.

If that is the case and knowing changing to 4 was for memory constraints, maybe writing CONTROL_RATE_PROFILE_COUNT to MSP is appropriate for allowing custom-define `CONTROL_RATE_PROFILE_COUNT=X` by pilot choice; otherwise default 4.  If written to MSP, then Configurator can programmatically set `numRateProfiles`.  -- just a thought, not a requirement.  I did see @ledvinap 's 😕 ermoji on PR.

ledvinap, 2025-02-06:
Maybe simplest way to handle this is to append CONTROL_RATE_PROFILE_COUNT to MSP_STATUS_EX.

closed 2025-02-06 by haslinghuis

nerdCopter, 2025-02-06:
All, using latest master (4.6-dev )you can now configure a custom define such as:
[image]

Note that each rate-profile seemingly costs about 112 bytes split unevenly between FLASH and RAM.

Thank you @haslinghuis , @ledvinap