betaflight/betaflight #14035: random chars in OSD usually after/during crashflip toggle.

nerdCopter, 2024-11-19:
### Describe the bug

[image]
https://youtu.be/_-bVHcVCoS0

non-blocking bug.

### To Reproduce

i think crashflip toggle, crashflip exit.

### Expected behavior

no random chars.

### Support ID

```txt
i'll produce one if needed. otherwise, build_id that i re-use locally: `e8ccfe0bdfbf8772b0ee0f8df43c3ce8`
```

### Flight controller

BETAFPVF4SX1280

### Add any other context about the problem that you think might be relevant here

This has been happening awhile using testing master or PR's based on 4.6. i do not if it happened in recall 4.5.x

haslinghuis, 2024-11-19:
Can you do a git bisect

https://github.com/betaflight/betaflight/commit/7156dc84a39f0068b45d6afa82ccd3afa6fc6cb6#diff-6268238c1b926c857b7cf783bfa5f4dba609904ff5cf565f6a58aa35989d6b43L26 was the latest change made:

nerdCopter, 2025-01-24:
closing the ticket unless i see it again. i have had no random characters in recent 4.6, so it might have been fixed with code refactoring. i did not re-investigate 4.5.x.

closed 2025-01-24 by nerdCopter