betaflight/betaflight #14037: Default cloud build 4.5.1 firmware for SpeedyBee F405 AIO is not functioning properly

MJ666, 2024-11-21:
### Describe the bug

I have updated my SpeedyBee F405 AIO with the latest firmware with BF configurator: 

# Betaflight / STM32F405 (S405) 4.5.1 Sep 19 2024 / 07:01:08 (77d01ba3b) MSP API: 1.46
# config rev: f3e29e7

The FC already came with BF 4.5.1 and the date was newer than this one (24th of September as far as i remember) and was flying fine with the same configuration before? I made a fresh cloud build with removing unused functions like OSD, LED, Acrotrainer and the Quad was flying well again:

# Betaflight / STM32F405 (S405) 4.5.1 Nov 21 2024 / 13:39:07 (77d01ba3b) MSP API: 1.46
# config rev: 79a4975

The symptoms I have seen with the bad build are oscillations and than motors going to full speed. From black box logs the oscillations look to be not induced by the gyro signals. Please see attached black box log and the configuration used.

[btfl_002.zip](https://github.com/user-attachments/files/17847085/btfl_002.zip)
[Settings SpeedyBee 135 4.5.1 all OTX.zip](https://github.com/user-attachments/files/17847095/Settings.SpeedyBee.135.4.5.1.all.OTX.zip)

### To Reproduce

Use the default cloud build version and try to fly. Not sure how my config will affect this behavior but there is nothing special even tuning is not finished yet.

### Expected behavior

Quad should fly normally.

### Support ID

```txt
2712d04b-eba2-40e4-aa57-5e1f83bd73e2
```

### Flight controller

SPEEDYBEEF405AIO

### Other components

Happymodel EX1404 4800KV motors
BetaFPV SuperD ELRS receiver
135mm AlienFlight custom frame
Tattu 650mAH 4S battery

### How are the different components wired up (including port information)

ELRS on UART6

MJ666, 2024-11-21:
Mhhh, the cloud version also has an outdated config? Should it not build a new firmware image if the config was updated?

BTW this is the version wich was preinstalled:

# Betaflight / STM32F405 (S405) 4.5.1 Sep 24 2024 / 14:34:30 (a8c404b64) MSP API: 1.46
# config rev: a8c404b64

nerdCopter, 2024-11-21:
dates are only compile date, nothing important. the "rev" is what is important.
`a8c404b64` is not an official build, or maybe it was a pull-request build. i.e. it is not in `4.5-maintnance`, nor `master` that i could find.
The `77d01ba3b` commit is dated `2024-07-26`; however there are also newer commits to `4.5-maintnance` for a future 4.5.2.
Consider building a pre-release 4.5.2:
[image]

nerdCopter, 2024-11-21:
@ot0tot , can you confirm/deny? this issue ticket alludes config file. and ticket owner does seem to have experience as per number of forked repos in his github.

MJ666, 2024-11-21:
a8c404b64 was the build the FC was deliverd with. Not sure how this build was created by SpeedyBee. The config is attached and working quite well already with a8c404b64 and the new cloud build i did with removed functions. PID's are not default and working well with other 135 builds. It is definitely not working with the defaut 4.5.1 firmware loaded with the configurator. The BBL file is from this non working firmware. At the very beginning you can see some motor output which is not caused be gyro or pid loop. Later the gyro is also reacting and motor output is saturatinga bit later. The Quad is starting to hover, than the oscilating is getting more havy and than it is shooting to the sky. The working and the non working firmware is build from the same commit id 77d01ba3b but with different features. Also the Gyro orientation needs to be fixed for the default cloud build version.

haslinghuis, 2024-11-21:
@MJ666 noticed gyro rate is very high. Can you check with 4.5.2 which is awaiting a AT32 bug to be fixed before release.

ot0tot, 2024-11-21:
> Also the Gyro orientation needs to be fixed for the default cloud build version.

This should be looked at. @kedeng can you check this?

MJ666, 2024-11-21:
> > Also the Gyro orientation needs to be fixed for the default cloud build version.
> 
> This should be looked at. @kedeng can you check this?

The actual config is fine. As already mentioned the default cached version on the cloud contains an outdated config.

nerdCopter, 2024-11-21:
> The actual config is fine. As already mentioned the default cached version on the cloud contains an outdated config.

@blckmn , please force update cloud-build `configs` cache

blckmn, 2024-11-21:
What is the build key with the problem?

MJ666, 2024-11-21:
New support id for the wrong build: 1cffd6d3-4ba5-4c89-b4e5-1dad9bf59773

MJ666, 2024-11-21:
> @MJ666 noticed gyro rate is very high. Can you check with 4.5.2 which is awaiting a AT32 bug to be fixed before release.

@haslinghuis Just updated to 4.5.2 and it is working as expected with the config posted.

haslinghuis, 2024-11-21:
Closing issue as non firmware but cloud build (cache) related.

closed 2024-11-21 by haslinghuis