betaflight/betaflight #14046: Launch mode OSD element missing HDZ AIO5

Justicefpv, 2024-11-26:
### Describe the bug

The launch mode OSD elements aren't working on the AIO5. Launch mode functions correctly, but the OSD elements that indicate it's enabled and display your launch angle aren't appearing.

### To Reproduce

Configure launch mode and use as intended.

### Expected behavior

OSD element should appear when launch mode is activated.

### Support ID

```txt
Id: "79ed5adf-1308-48bf-8580-69ae422ac56f"
```

### Flight controller

HDZ AIO5 (CRAZYBEEF4SX1280)

haslinghuis, 2024-11-26:
Don't have any HDzero hardware, could it be the font ?

```bash
set displayport_msp_fonts = 0,0,0,0
```

Justicefpv, 2024-11-27:
> Don't have any HDzero hardware, could it be the font ?
> 
> ```shell
> set displayport_msp_fonts = 0,0,0,0
> ```

I will check this evening, but this is not working as they come out of the box on 4.4.2 and 4.6 master builds. I've not made any changes to the default font.

Justicefpv, 2024-11-27:
@haslinghuis By default, the following was set `displayport_msp_fonts = 0,1,2,3` 

I changed it to `displayport_msp_fonts = 0,0,0,0` but sadly, this doesn't resolve the issue.

haslinghuis, 2024-11-27:
@SteveCEvans pls have a look

SteveCEvans, 2024-11-27:
Away for a few days. I’ll take a look when I return home.

SteveCEvans, 2024-12-18:
It works, but note the following:

1. Be sure to enable warnings on the OSD tab.
[image]
2. Build with `LAUNCH_CONTROL` defined.
[image]

Justicefpv, 2024-12-18:
@SteveCEvans I feel rather foolish and apologise for taking up your time, I had indeed not enabled warnings, nor had the people who I also had checked this.

closed 2024-12-18 by Justicefpv

SteveCEvans, 2024-12-18:
No need to apologise; it took me a while to realise I'd not enabled them either!