betaflight/betaflight #14059: GPS

sky30432, 2024-12-01:
### Is your feature request related to a problem? Please describe

GPS

### Describe the solution you'd like

 
# Building AutoComplete Cache ... Done!
# 
# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.28V, Core temp=46degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 3826, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=ICM42688P, ACC=ICM42688P, BARO=DPS310
OSD: MSP (30 x 13)
BUILD KEY: 3f5a02b642f42a98a81eca2520471cd5 (4.5.0)
System Uptime: 40 seconds, Current Time: 2024-12-01T22:13:50.148+00:00
CPU:39%, cycle time: 123, GYRO rate: 8130, RX rate: 15, System rate: 9
Voltage: 1 * 0.01V (0S battery - INIT)
I2C Errors: 0
SD card: Not configured
FLASH: JEDEC ID=0x00ef4017 8M
GPS: NOT CONNECTED, UART4 19200 (set to 57600), NOT CONFIGURED, version =  unknown
Arming disable flags: RXLOSS CLI MSP RESCUE_SW DSHOT_TELEM

### Describe alternatives you've considered

 
# Building AutoComplete Cache ... Done!
# 
# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.28V, Core temp=46degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 3826, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=ICM42688P, ACC=ICM42688P, BARO=DPS310
OSD: MSP (30 x 13)
BUILD KEY: 3f5a02b642f42a98a81eca2520471cd5 (4.5.0)
System Uptime: 40 seconds, Current Time: 2024-12-01T22:13:50.148+00:00
CPU:39%, cycle time: 123, GYRO rate: 8130, RX rate: 15, System rate: 9
Voltage: 1 * 0.01V (0S battery - INIT)
I2C Errors: 0
SD card: Not configured
FLASH: JEDEC ID=0x00ef4017 8M
GPS: NOT CONNECTED, UART4 19200 (set to 57600), NOT CONFIGURED, version =  unknown
Arming disable flags: RXLOSS CLI MSP RESCUE_SW DSHOT_TELEM

haslinghuis, 2024-12-06:
Hard to investigate without knowing GNSS units and protocol used here.

haslinghuis, 2024-12-12:
https://content.u-blox.com/sites/default/files/products/documents/NEO-6_DataSheet_%28GPS.G6-HW-09005%29.pdf

- Try a lower `gps_update_rate_hz`
- Try with auto config disabled.
- Baudrate needs to match with u-center

rizacelik, 2024-12-12:
When I checked the GPS module with u-center, I saw that the baud rate was 230400. I changed the baud rate value to 57600 with u-center and it worked.

Thank you for your help.

```
Entering CLI Mode, type 'exit' to reboot, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.29V, Core temp=45degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 4366, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1
GYRO=MPU9250, ACC=MPU9250, MAG=AK8963
OSD: MSP (53 x 20)
System Uptime: 3207 seconds, Current Time: 2024-12-12T21:03:31.039+00:00
CPU:51%, cycle time: 125, GYRO rate: 8000, RX rate: 0, System rate: 9
Voltage: 893 * 0.01V (3S battery - CRITICAL)
I2C Errors: 1
SD card: Not configured
GPS: connected, UART1 57600 (set to 57600), auto config OFF, version =  M6
Arming disable flags: RXLOSS ANGLE LOAD CLI MSP
```

closed 2025-01-20 by github-actions[bot]