betaflight/betaflight #14062: Unable to calibrate magnetometer in betaflight 4.5.1

Patronics, 2024-12-04:
### Describe the bug

When attempting to calibrate the magnetometer, there's no feedback from the drone indicating whether calibration is starting, and after rotating all axes as instructions describe, calibration values remain 0,0,0.

The drone doesn't have a dedicated beeper, but it does have dshot beacons enabled (it's unclear whether those are used or not for these calibration beeps). There is no beep from the dshot beacon to indicate if calibration is actually starting or not. All beeper-related switches in the configuration tab are enabled (aside from the "tx lost" setting, which is not relevant to this issue). The "Calibrating" Arm Disable flag does show up when pressing the calibrate magnetometer button, indicating that it is at least starting the calibration.

The drone is also large and heavy, so I might not be successfully shaking it at the 350° Per second rate that the calibration instructions indicate. After shaking it, and spinning it on all axes, there's still no feedback indicating if calibration was successful, unsuccessful, or never started at all. Checking with the `get mag` command in the CLI simply returns 0,0,0 for the calibration values.

I've tried starting the calibration from betaflight configurator (through a USB extension cable to avoid interference from some magnets on the back of my laptop's lid), as well as through the controller's sticks, and through the speedybee app over bluetooth. The results are the same in all 3 cases. 

The magnetometer is definitely connected and I can see valid data from the debug page for it, which clearly changes when rotating the drone or moving magnets around near it, so that's not the problem. 

### To Reproduce

Attempt to start a calibration of the magnetometer. Observe that no feedback is given, and that the drone does not save any changes to calibration settings after moving it as instructed.

### Expected behavior

As described in the instructions, the beeper should beep to provide feedback, and the drone should save changed calibration values after being spun on each axis. 

### Support ID

```txt
3bcce176-3dc5-4334-9418-3dc616521122
```

### Flight controller

SpeedyBee F405 V4

### Other components

HGLRC M100-5883 GPS/Compass unit (with magnetometer QMC5883 connected via i2c)
Speedybee ELRS reciever. 
DJI Air unit and V1 FPV Goggles
2 SpeedyBee 4-in-1 50A ESCs, each running BlueJay 0.21.0

Patronics, 2024-12-04:
After doing a bit more research, perhaps issue #13338  is relevant? Though for those users it seemed to be resolved by lowering the threshold speed to 350° per seconds (from 450), that might not be adequate for such a large drone as this one (with a several foot wingspan), which I'm worried could be damaged by shaking it too violently. Also as mentioned there's no beeping or other clear feedback of progress, despite all beeper switches being enabled.

ctzsnooze, 2024-12-06:
For mag cal, motors don’t beep, only a real beeper works.
Unless the mag cal threshold is exceeded, the cal won’t start.
The gyro rate can be checked using the sensors tab.  Be sure to exceed that rate to start the calibration.

ctzsnooze, 2024-12-17:
Were you able to get it to work @Patronics ?
The LED on the FC does change behaviour during calibration, and a Beeper is very useful indeed for Mag calibration at the field.
If you use the Sensors tab to show gyro rate, you should be able to work out something that induces a transient at 350 deg/s.
Normal handing of a 5in quad readily exceeds 350 deg/s; to not exceed 350 deg/s requires lifting the aircraft with some care.
Anyway let me know if you have got to a reasonable conclusion here, please.

Patronics, 2024-12-18:
Hey, thanks for reaching out to check in. I did confirm from the sensors tab that the gyro rate exceeded 350 degrees per second, but still haven't had any success with the calibration. I didn't notice any change in LED patterns, and tried several times with the magnetometer configured for different orientations (since I'm not certain which orientation is correct on this GPS module), but the calibration values always came back as 0,0,0. 

For the moment I've disabled the magnetometer so I can fly the octocopter without it, but happy to try again if you have any more ideas to try. 

Also I'm curious, is there any reason that the motor beacons aren't used in addition to a dedicated beeper for this calibration? I could see them potentially causing interference with the magnetometer, but presumably not substantially, since the motors are of course also active during flight.

aowi7280, 2025-01-08:
Just set the cal values to 1,1,1. If you have the orientation of the Mag set correctly, it will be close enough to work.

Patronics, 2025-02-08:
> Just set the cal values to 1,1,1. If you have the orientation of the Mag set correctly, it will be close enough to work.

Sorry for the late reply, I've been busy and didn't see this message, but I'll give this a try next time I have a chance to go flying. Thanks for the suggestion!

closed 2025-03-18 by github-actions[bot]