betaflight/betaflight #14066: VTX: SmartAudio 2.0 Pitmode issue with Eachine TX1200

menschel, 2024-12-06:
### Describe the bug

The VTX does not enter / use pit mode on request.
In Configurator on the VTX page, the pit mode switch does nothing.
Setting up a range for AUX switch to actuate pit mode does nothing as well.

### To Reproduce

On a Setup with Eachine TX1200 VTX, try to enter pit mode via Configurator.
Observe that it's not entering pit mode, i.e. the green power leds don't start to blink.

### Expected behavior

Device enters and exits pit mode as requested.

### Support ID

```txt
f448fe58-a5a8-4207-a7b4-4f78d6768505
```

### Flight controller

Kakufe F4 V2.4

### Other components

- Holybro M10 GPS
- Eachine TX1200 VTX -> Smart Audio 2.0
- Betafpv Nano Rx 2G4 - ELRS V3.5.3
- 4x Holybro Tekko32 F4 Single ESCs w AM32 V2.16

### How are the different components wired up (including port information)

ELRS -> Serial6
GPS -> Serial4
VTX -> Serial3
ESCs - M1-4

### Add any other context about the problem that you think might be relevant here

This behavior has a lot in common with a previously reported issue regarding the TX805 PR https://github.com/betaflight/betaflight/pull/10436

The only way to enter pit mode is to go via OSD menu.
The pit mode option in OSD menu does nothing, but it the workaround mentioned in the PR above works does work.

In OSD menu
- FEATURES->VTX->PIT does nothing
- FEATURES->VTX->CONFIG->"OP MODEL" set to "RACE"
- FEATURES->VTX->CONFIG->"PIT FMODE" set to "PIR" (resets to "---" if pit mode is left)

After that HW indicates that pit mode is active but the indication in Configurator shows that pit mode is off.

Pit mode is left on actuation of AUX switch.

nerdCopter, 2024-12-06:
does the VTX actually have pit mode? i thought only Tramp protocol had pitmode.  your VTX minimum is reported to be 25mW.
[image]

menschel, 2024-12-06:
> does the VTX actually have pit mode? i thought only Tramp protocol had pitmode. your VTX minimum is reported to be 25mW. [...]

Yes it has pit mode. It is indicated by all 3 green power leds blinking.

I have started digging, and IMO there are some conditional evaluations that exclude SA 2.0 from pit mode features.

[This](https://github.com/betaflight/betaflight/blob/master/src/main/cms/cms_menu_vtx_smartaudio.c#L150) seems to be the reason why I see "---" on PIT FMODE each time, so this is not the issue but it is able to set it afterwards.

[This](https://github.com/betaflight/betaflight/blame/master/src/main/io/vtx_smartaudio.c#L680) compare with SA 2.1 is also fishy.

Git Blame leads to https://github.com/betaflight/betaflight/pull/8380 but I'm not sure if it's the cause or just a relocation.

menschel, 2024-12-06:
That early return may be the reason for my issue.
```
    if (onoff && saDevice.version < 3) {
        // Smart Audio prior to V2.1 can not turn pit mode on by software.
        return;
    }
```
https://github.com/betaflight/betaflight/blob/master/src/main/io/vtx_smartaudio.c#L931

menschel, 2024-12-06:
LOL, ardupilot does not care about Version and just writes it.
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_VideoTX/AP_SmartAudio.cpp#L201

Just reading

[SmartAudio 2.1 dropped support for outband pitmode so we won't support it](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_VideoTX/AP_SmartAudio.cpp#L534)

Is it a valid test for the Protocol Version if I try to set POR pitmode on the VTX,
i.e. Is it proof that protocol is SA2.1 and not SA2.0 if I cannot switch into POR pitmode?

menschel, 2024-12-06:
OK, I tried to switch to POR pitmode via OSD menu and it failed. Is that proof that the VTX is actually SmartAudio 2.1 and falsely advertises itself as SmartAudio 2.0?

menschel, 2024-12-18:
After asking Eachine Support, they confirmed that this VTX uses SA 2.0 protocol.
Not sure how to handle this now. Please advise.

nerdCopter, 2024-12-18:
old video and different model by Eachine, but maybe useful. TLDW set VTX to race-mode so that it boots into pit-mode always.
Maybe this no longer works with VTX-Tables, i don't know.
https://www.youtube.com/watch?v=VuZC4RPxfig

menschel, 2024-12-18:
OK, in general I can cope with the behavior.
As mentioned on top, the workaround correctly selects pitmode on boot.

The things that don't work are pitmode on AUX switch (I use the inverse of ARMING),
and the slider button in the configurator.

I'll do some flights and then decide if I close this Issue or push a PR to fix it.

closed 2025-01-25 by github-actions[bot]