betaflight/betaflight #14076: the "X" icon to delete in Modes tab not visible on dark mode

jonathanoeijoeng, 2024-12-09:
### Is your feature request related to a problem? Please describe

The "X" icon to delete on Modes tab not visible on dark mode 

Icon on dark mode
[image]

Icon on light mode
[image]

### Describe the solution you'd like

Change x icon to white to make it more visible

### Describe alternatives you've considered

I think it would be much better if we can change icon to white to make it more visible on dark mode

nerdCopter, 2024-12-09:
@VitroidFPV

VitroidFPV, 2024-12-17:
Apologies that I didn't get to this earlier, though it should've been made in the Configurator repo 😅