betaflight/betaflight #14084: 4.6.0 cli init

NightCrack, 2024-12-11:
### Describe the bug

Can't initialize  cli with "#" via command.
I'm using screen on Ubuntu; the line is as follows:

`screen -S $SCREEN_NAME -p 0 -X stuff "#^M"`

The screen session exists, created by current user; SCREEN_NAME value is valid. If I enter the screen session in question and input "#" manually cli successfully activates. After that, all commands sent via method described above are valid and working; the issue occurs only on initiation stage.

This issue persist with betaflight 4.6.0; with 4.5.1 everything works ok.

### To Reproduce

Create `screen` session for betaflight 4.6.0 fc connection and send `"#^M"` command via `-X stuff` subcommand.

### Expected behavior

betaflight cli being successfully initiated so the further interaction may be possible.

### Support ID

```txt
a368eb5d-8361-4947-8a15-f3362f3f4e36
```

### Flight controller

AOCODARCF405V3

ledvinap, 2024-12-11:
`screen -S $SCREEN_NAME -p 0 -X stuff "#"`

There must be delay after `#`, otherwise the command is cancelled. This is 'wanted side effect', there is some discussion in `PORT_CLI_CMD` PR

NightCrack, 2024-12-11:
How long this delay must be exactly? I had no response in about a minute of waiting. Again, the behavior is drastically different with 4.5.1 version firmware.

haslinghuis, 2024-12-11:
https://github.com/betaflight/betaflight/commit/c2900de5c10545955b0798450b39914ea62e91b1

ledvinap, 2024-12-11:
The delay should be about 10 ms (serial is checked on 100Hz, only `#` must be received during check).
But I'd like to add larger guard time in future, so you can use 1s (it will be probably less) both before and after '#' to be future-proof ..

NightCrack, 2024-12-11:
> both before and after

I can understand delay after the command (from my earlier experience), but before the command - what for?

Also I use the next loop to assure cli mode initiation and shorten time delay to a minimum necessary:

` # Check for dump end line to proceed
 N=1
 while [ $N -ne 0 ]; do
  grep "save configuration" $BETAFLIGHT_LOG
  N=$?
 done`

ledvinap, 2024-12-12:
> I can understand delay after the command (from my earlier experience), but before the command - what for?

To make it more unlikely that random character is recognized as CLI entry.

Maybe https://github.com/betaflight/betaflight/pull/13940 will be better for your use - you get better command/reply separation, don't have to filter prompts etc ..

nerdCopter, 2024-12-12:
@ledvinap , his support id shows `master` @ `aab21384f` which is the latest commit as of the moment. as the PR mentioned is Oct 3, it already contains such.  or am misunderstanding and a new client-method is required?

@NightCrack , you might consider flashing the commit just prior to the mentioned PR's merge:
```
c2900de5c	J Blackman	2024-10-03	Adding CLI command pass through for MSP (#13940)
23e682c31	Mark Haslinghuis	2024-10-03	Fix build when using SRXL telemetry (#13947)
```
so maybe compare your findings against the old code at `23e682c3104c98406d5eae8dbc3ee5e445c5041b`
[image]

this will help if a code-solution is needed.

ledvinap, 2024-12-12:
@nerdCopter :  CLI command pass through for MSP (#13940) changed CLI enter timing  a bit. This was intentional change, not a bug. @NightCrack is sending two characters, '#\n', this won't work. But sending just '#'  should work fine.

And  #13940 may be easier to use for scripting CLI.

NightCrack, 2024-12-12:
To be honest, I don't follow you on the provided suggestions: I can't find the way to select the exactly needed commit for betaflight firmware (I use beta flight configurator, stable/web). Furthermore, I've tried to form the command in a suggested way (`[STX]#[LF][ETX]`), with hex and ctrl variations of it—none have been completed successfully. I only received `` as an output.

But, I've found a workaround for my case: following the ChatGPT suggestion

> 
[image]

I've tried the -fn option and later modified my whole session setup with it:
`screen -dmSL $SCREEN_NAME -Logfile $BETAFLIGHT_LOG -fn $DEVLINK 115200

Can't say that the issue was resolved, I'd rather that there was more documentation on such topics, with some bare examples.

haslinghuis, 2024-12-12:
- documentation is mostly an after thought - maybe even more in open source
- feel free to add documentation - as this feature is for advanced usage of the firmware and just recently added.

nerdCopter, 2024-12-12:
so if the problem is `SCREEN` and not betaflight, please close this ticket.
to answer on flashing the commit, this screenshot; however it seems this test is no longer needed.
[image]

NightCrack, 2024-12-12:
@nerdCopter 
The problem is not a `SCREEN` or betaflight itself, but rather the way of interaction with betaflight.
What my feedback shows is that there is a way to completely ignore the dev's attempts to do a flow control.

Either way, this topic helped me all the way. Thanks for showing the way to select commit, I may try it later.

closed 2024-12-12 by NightCrack

ledvinap, 2024-12-12:
> Furthermore, I've tried to form the command in a suggested way (`[STX]#[LF][ETX]`), with hex and ctrl variations of it—none have been completed successfully. I

It is now how it works - after #13940, you don't have to enter CLI mode ('#'). You can intermix MSP and CLI commands.

You should send `[STX]<command>[LF][ETX]` and get `[STX]<command response>[ETX]` reply. `#` is IMO ignored as command, you you see is `[STX][EXT]` (try passing output through hexdump -c)

try `echo -e "\x02help\n\x03"`