betaflight/betaflight #14098: conflict between motors outputs and strip LED

airbotsystems, 2024-12-19:
### Describe the bug

When Led strip function is active, it works well but do not allow to access the motor page without getting an error message. Motors output are inactives.
After turning off the led strip function, motor page load normal, without error popup essga and all MOT outputs are functional.

### To Reproduce

hard because it is on a custom board, actually in dev
I suppose a conflict between dmas but not sure

### Expected behavior

Led and motors working well independentely

### Support ID

```txt
d5b49aca-2177-4cfe-9aeb-01f910e90061
```

### Flight controller

Custom H7 FC

haslinghuis, 2024-12-19:
timer conflict

airbotsystems, 2024-12-19:
LED strip is A8 and is the only on TIM1
[image]