betaflight/betaflight #14102: No custom defaults

dyeballer77, 2024-12-24:
### Describe the bug

When updating firmware on matekf411 board any target above 4.3.2 doesn't contain custom defaults, and the accelerometer and gyro arn't recognized.

### To Reproduce

Update a matekf411 board from 4.3.2 to anything higher and the problem will be obvious.

### Expected behavior

I expected to update the firmware, get a message to apply custom defaults on first connection and the accelerometer and gyro to be recognized. The support ID was sent from a working configuration of betaflight 4.3.2 any version of betaflight above 4.3.2 is where the problem presents itself...

### Support ID

```txt
d3f99359-daf0-4068-a239-e7c5e0501dd4
```

### Flight controller

matekf411 it came with my tyro79 drone

YarosMallorca, 2024-12-24:
I am not a firmware developer, but I'm pretty sure that Custom Defaults are applied automatically.
Maybe this is an issue with the target?

haslinghuis, 2024-12-24:
See: https://github.com/betaflight/unified-targets/issues/1119#issuecomment-1683989184

closed 2024-12-24 by haslinghuis

dyeballer77, 2024-12-25:
> See: betaflight/unified-targets#1119 (comment)

it is definitely a problem with the target.....is there a way to apply the
custom defaults manually?.....everything works fine up to BF 4.3.2 and then
you guys changed the way the targets are compiled, or atleast thats how it
looks, and thats where the problem shows itself......

dyeballer77, 2024-12-25:
I found this in the unified target for matekf411

#define USE_GYRO
#define USE_GYRO_SPI_MPU6000
#define USE_ACC
#define USE_ACC_SPI_MPU6000
#define USE_BARO
#define USE_BARO_BMP280
#define USE_MAX7456

These are the custom defaults correct??? If yes, can you please tell me
step by step how to enter them, I'm guessing its is done through CLI, but i
have copy and pasted and it doesn't seem to work. Can you please tell me
exactly what Im suppose to do and type, it would be greatly appreciated, I
love learning the ins and outs of this hobby, this one has got me
stumped....

dyeballer77, 2024-12-25:
Nevermind I GOT IT WORKING!!!!.....what a nightmare...I don't understand
why they would make it so difficult moving from one version of BF to
another, seriously, it's like they are trying to confuse and anger
people....