betaflight/betaflight #14111: IFLIGHT F745 V1 Gyro Post 4.4

mzirkle44060, 2024-12-27:
### Describe the bug

Lost the Gyro, trying to flash my FC from master.
Core dump ID: 1fd324af-c99e-4069-837d-0a35dea445f3

### To Reproduce

Flash the FC from 4.3 to to 4.4

### Expected behavior

Gyro should work, looks like in the master branch, the USE_ACC_SPI_MPU6000 define in the config, but manufacturer states the following...
Gyro: ICM42688

### Support ID

```txt
1fd324af-c99e-4069-837d-0a35dea445f3
```

### Flight controller

https://www.getfpv.com/iflight-blitz-f7-v1-1-aio-board-f745-fc-55a-2-6s-bluejay-esc-25x25.html

ot0tot, 2024-12-29:
The "bug" is that the manufacturer is incompetent. They are well aware of the rules and the proper procedure for changing targets. You should hold them to this.

mzirkle44060, 2024-12-29:
I get that, ironically, I fried 2 of the ESC chips, so in the Binn it goes
:) thx mate

mzirkle44060, 2024-12-30:
The "Future Proof" board, turns out is not so Future proof! Wasn't a cheap investment either!

closed 2024-12-30 by mzirkle44060

nerdCopter, 2025-01-02:
in case it is not clear, you can add custom defines for missing gyro:
[image]
`ACC_SPI_ICM42688P GYRO_SPI_ICM42688P`

mzirkle44060, 2025-01-02:
I was in the process of testing that, but it looks as if this BetaFlight define eluded me... However, the 2 of the motor ports on the AIO somehow got smoked thus ending the fun ;(

image.png (view on web)<https://github.com/user-attachments/assets/25a0f3fa-ef8a-4dab-99bc-6191d5dbab68>