betaflight/betaflight #14118: STM32G474CEU6 uses gyroscope 42688 and the 3D model cannot be rotated on Betaflight Configurator.

LYNHQQ, 2024-12-31:
### Describe the bug

After G474 was connected to 42688, the gyroscope data was confused in the Configurator, and the 3D model was at rest. When I switched to BMI270, it worked normally
[image]
[image]

### To Reproduce

Enter the same firmware and it will appear，When I switched 42688 to BMI270, it worked fine, I confirmed that 42688 was good, it was switched from another flight controller that worked fine

### Expected behavior

It should be normal for 42688 to work on G474

### Support ID

```txt
###

# # Problem description

# #

# ###

# version
# Betaflight / STM32G47X (SG47) 4.5.1 Dec 31 2024 / 09:09:19 (77d01ba3b) MSP API: 1.46
# config rev: 5a85a4e
# board: manufacturer_id: TEST, board_name: 474

# status
MCU G474 Clock=168MHz (PLLR-HSI), Vref=3.29V, Core temp=43degC
Stack size: 2048, Stack address: 0x2001fff0
Configuration: CONFIGURED, size: 3675, max available: 8192
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1
GYRO=ICM42688P, ACC=ICM42688P
OSD: MSP (53 x 20)
System Uptime: 768 seconds, Current Time: 2024-12-31T02:08:25.405+00:00
CPU:58%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
Arming disable flags: RXLOSS CALIB CLI MSP

# flash_info
Flash sectors=0, sectorSize=0, pagesPerSector=0, pageSize=0, totalSize=0 JEDEC ID=0x00000000
Partitions:
FlashFS size=0, usedSize=0

# dump master

# version
# Betaflight / STM32G47X (SG47) 4.5.1 Dec 31 2024 / 09:09:19 (77d01ba3b) MSP API: 1.46
# config rev: 5a85a4e

# start the command batch
batch start

board_name 474
manufacturer_id TEST

# name: -

# resources
resource BEEPER 1 B07
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A06
resource MOTOR 4 A07
resource MOTOR 5 NONE
[... 988 lines cut ...]
resource MOTOR 3 A06
resource MOTOR 4 A07
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 C10
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 C11
resource I2C_SCL 2 C04
resource I2C_SDA 2 A08
resource LED 1 B02
resource LED 2 A03
resource SPI_SCK 1 B03
resource SPI_SCK 2 B13
resource SPI_SDI 1 B04
resource SPI_SDI 2 B14
resource SPI_SDO 1 B05
resource SPI_SDO 2 B15
resource ADC_BATT 1 A00
resource ADC_CURR 1 A01
resource FLASH_CS 1 B06
resource GYRO_EXTI 1 C13
resource GYRO_CS 1 B12

haslinghuis, 2024-12-31:
- Most probably the board design has not anticipated necessary design requirements for the ICM42688P.
- AirbotG4AIO works fine with ICM42688P.

closed 2024-12-31 by haslinghuis

ot0tot, 2025-01-13:
There seems to be some issue with the SPI2 clock on the G473. Try compiling with ` -D'ICM426XX_CLOCK=6000000`. The issue is being investigated.