betaflight/betaflight #14125: 4.6.0-dev | OSD Menu Missing on https://app.betaflight.com/# | defaults nosave

bmsleight, 2025-01-04:
### Describe the bug

OSD menu tab is not available on https://app.betaflight.com/# 

When defaults nosave is entered. Commenting out this line, reset setting then loading the file OSD menu tab is shown.  

### To Reproduce

Using cli:-

```
defaults nosave 
save
```

Goto > Set-up -> Reset Setting 
https://app.betaflight.com/#

CLI -> Load from File -> Execute

Connect - Then OSD tab is missing

Edit diff dump, comment out defaults nosave 
(Important to reset setting) 

Goto > Set-up -> Reset Setting 
https://app.betaflight.com/#
CLI -> Load from File -> Execute

Connect - Then OSD tab is present.

```
$ diff Jessica_BTFL_cli_JESSICA_20250104_122458_FLYWOOF405PRO.txt OSD_Working_Jessica_BTFL_cli_JESSICA_20250104_122458_FLYWOOF405PRO.txt 
13c13
< defaults nosave
---
> # defaults nosave

```

[Jessica_BTFL_cli_JESSICA_20250104_122458_FLYWOOF405PRO.txt](https://github.com/user-attachments/files/18307253/Jessica_BTFL_cli_JESSICA_20250104_122458_FLYWOOF405PRO.txt)
[OSD_Working_Jessica_BTFL_cli_JESSICA_20250104_122458_FLYWOOF405PRO.txt](https://github.com/user-attachments/files/18307254/OSD_Working_Jessica_BTFL_cli_JESSICA_20250104_122458_FLYWOOF405PRO.txt)

### Expected behavior

OSD Menu 

[image]
[image]

### Support ID

```txt
###

# # Problem description

# # 4.6.0-dev | OSD Menu Missing on https://app.betaflight.com/# | defaults nosave

# ###

# version
# Betaflight / STM32F405 (S405) 4.6.0 Jan  4 2025 / 12:42:07 (f6985a8bf) MSP API: 1.47
# config rev: 9905ee9
# board: manufacturer_id: FLWO, board_name: FLYWOOF405PRO

# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.26V, Core temp=43degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 3670, max available: 16384
[... 1067 lines cut ...]
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
    CH2 : DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

#
```

### Flight controller

FLWO, board_name: FLYWOOF405PRO

### Add any other context about the problem that you think might be relevant here

[Submit Support Data.txt](https://github.com/user-attachments/files/18307257/Submit.Support.Data.txt)
[Submit Support Data with OSD Showing.txt](https://github.com/user-attachments/files/18307258/Submit.Support.Data.with.OSD.Showing.txt)

haslinghuis, 2025-01-04:
- Build key shows `OSD_HD` but feature OSD is disabled.
- Is OSD disabled in configuration tab ?
- Try enabling `feature OSD` in CLI

bmsleight, 2025-01-04:
>Is OSD disabled in configuration tab ?
Yes  (Sorry did not look there!)

>Try enabling feature OSD in CLI
This resolve and menu tab now available. 

Was upgrade from 4.4.3 to 4.6.0-dev.

My 4.4.3 - diff all did not contain
`feature OSD`

When I do a reset all on 4.6.0-dev, then a diff all it contains  feature OSD
```
# reset configuration to default settings
defaults nosave

...

# feature
feature OSD

```

Thanks for quick response.

bmsleight, 2025-01-04:
Mark as closed use error on upgrade from 4.4.2 to 4.4.3 to 4.6.0-dev ?

closed 2025-02-11 by github-actions[bot]