betaflight/betaflight #14149: 4.5.1 -> 4.5.2 battery voltage in the OSD breaks communication with HDZero vtx

XaocuHKa, 2025-01-12:
### Describe the bug

after upgrading to 4.5.2 if "battery voltage" is checked in the OSD tab, the whole OSD stops working, sometimes was getting an error "motor_protocol" when trying to arm.

### To Reproduce

flash 4.5.2, check "battery voltage" in the OSD tab

### Expected behavior

it is expected for the "battery voltage" to show value on the FPV feed from the VTX.

works fine on 4.5.1

### Support ID

```txt
Id: "4b00a9c7-d46f-49ee-9f69-2aea7ec0495d"
```

### Flight controller

TBS_LUCID_PRO

### Other components

VTX is HDZero V3

ot0tot, 2025-01-12:
Have you tried with a 4kHz PID loop rate and DSHOT300 motor protocol?

XaocuHKa, 2025-01-12:
> Have you tried with a 4kHz PID loop rate and DSHOT300 motor protocol?

just tried it. Same behavior with 4khz PID loop, and same with 4khz PID loop + DSHOT300. Disabling Bidirectional DSHOT doesnt help either.

limonspb, 2025-01-12:
I was `git bisecting` and @XaocuHKa was flashing and re-testing. Thanks Din!

Found out it's this PR caused the problem in 4.5-maintenance:
https://github.com/betaflight/betaflight/pull/13839
which is a cherry-pick from this:
https://github.com/betaflight/betaflight/pull/13813
@SteveCEvans  @haslinghuis 

```
aa81b811c4dfbbb9d28fe9affde2293a1a5b26ae is the first bad commit
commit aa81b811c4dfbbb9d28fe9affde2293a1a5b26ae
Author: Mark Haslinghuis <...>
Date:   Fri Aug 30 16:45:31 2024 +0200

    [4.5.2] Split OSD element rendering into draw and display states (#13813) (#13839)    
```

SteveCEvans, 2025-01-13:
Any more hints on how to reproduce? Testing using an ORBITF435 and HDZ Freestyle V2 I'm seeing a battery voltage of 12.9V displayed without issue. FC to VTX communication continues without a hitch.

This is using the `betaflight/4.5-maintenance` branch, 3cd5de187a32950d5b3e7c9eaea39cbfaa5d984a

[image]

Something interesting of note here is that we're wasting bytes send a command to write an empty warning string at row 15, column 19. 

Any suggestions of critical factors to reproduce.

XaocuHKa, 2025-01-13:
[BTFL_cli_1_20250113_153806_TBS_LUCID_FC.txt](https://github.com/user-attachments/files/18403327/BTFL_cli_1_20250113_153806_TBS_LUCID_FC.txt)
maybe try that dump?

SteveCEvans, 2025-01-14:
Tried that, with all the FC specific resources, bus assignments, pinio config etc. removed. Unfortunately I don't have this exact FC.

limonspb, 2025-01-14:
> [BTFL_cli_1_20250113_153806_TBS_LUCID_FC.txt](https://github.com/user-attachments/files/18403327/BTFL_cli_1_20250113_153806_TBS_LUCID_FC.txt) maybe try that dump?

Can you ask Trappy to send a couple of FCs to Steve?

SteveCEvans, 2025-01-19:
Right, I have the FC. Using the config from the support ticket, with the setup below; ie no RX or ESC I don't see any issue still.

[image]

So it looks like some more information is needed about the setup.

When the OSD hangs, is it just the OSD that has the issue? With the configurator connected does everything still work as expected on the receiver tab? Can the motors be commanded to spin on the motors tab?

XaocuHKa, 2025-01-20:
hey Steve, made a video: https://youtu.be/ROxt6hibrX0?si=c2sIn9wMAfOewdQs
can you please check it out?

When the OSD hangs, is it just the OSD that has the issue? - yes/no, i've had some issues with arming but now icant reproduce
With the configurator connected does everything still work as expected on the receiver tab? - yes
Can the motors be commanded to spin on the motors tab? - yes

i'll update the thread with my other quad having the same issue on Foxeer V4

SteveCEvans, 2025-01-20:
Can you please confirm your build options, just in case the support ID misses something.

[image]

Also, why is your reception so bad on the monitor? Are you in pit mode?

SteveCEvans, 2025-01-20:
I've rebuilt as per the support ID using the below and I now see the failure.

```
make TBS_LUCID_FC EXTRA_FLAGS="-DCLOUD_BUILD -DUSE_ACRO_TRAINER -DUSE_DSHOT -DUSE_GPS -DUSE_GPS_PLUS_CODES -DUSE_LED_STRIP -DUSE_OSD -DUSE_OSD_HD -DUSE_OSD_SD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_VTX"
```

SteveCEvans, 2025-01-20:
Neither a `MATEKF405TE` nor an `AIRBOTSUPERF4` (AT32) show this issue with the same build config.

SteveCEvans, 2025-01-21:
This issue was caused by a slow ramp on the voltage ADC reading, peculiar to the TBS_LUCID_FC which resulted in it attempting to flash the battery voltage whilst rendering the Betaflight logo. It should have just skipped past the rendering routine for the battery voltage, but because it was in the off blink state the OSD state machine stalled waiting for the rendering to complete, which never happened and the OSD stalled.

This fix sets the default state to an element to be rendered so if an element is flashing and is in the off state it will be skipped. This may have been causing OSD issues when any element was flashing as rendering would stall when any element was in the off blink state. This won't have stalled the FC, but will have resulted in an increased load in the OSD task and possible stalls of the OSD synchronous with the blink rate.

XaocuHKa, 2025-01-21:
I've flashed the commit #14188, tested and the issue is not there anymore.

blckmn, 2026-09-27:
This has since been addressed in Betaflight, so we're closing it as part of a clean-up of issues more than a year old. If the latest release doesn't cover what you were after, please open a new issue and link back here.

closed 2026-09-27 by blckmn