betaflight/betaflight #14164: Betaflight 4.6 - Position Hold - OSD Mode not Change

iStitch07, 2025-01-15:
### Describe the bug

During testing the new Altitude Hold and Position Hold features, I noticed that when switching to the Position Hold mode, the current flight mode is displayed as ALTH. I assume it should be POSH

### To Reproduce

To reproduce: assign Altitude Hold to one position of a three-position switch and Altitude Hold + Position Hold to another position (based on the guide from the wiki - https://betaflight.com/docs/wiki/guides/current/position-hold-v4-6). First, switch to Altitude Hold mode, then switch to Altitude Hold + Position Hold mode. The OSD will continue to display ALTH. It’s possible that this mode has “priority” in the display and “overrides” the Position Hold mode indication.

### Expected behavior

It is expected that the Position Hold mode will be displayed on the OSD as POSH.

### Support ID

```txt
515c3294-12e8-43a7-98c4-69a61bf11cd2
```

### Flight controller

FLYWOOF405PRO

### Add any other context about the problem that you think might be relevant here

Perhaps I misunderstood the information from the wiki, and these modes are not supposed to be activated simultaneously but rather assigned to different positions on the same switch.

nerdCopter, 2025-01-15:
@ctzsnooze

haslinghuis, 2025-01-15:
Should flight mode `POSH` take precedence over `ALTH` ?

PR #14165 (2025-01-17): Flip FM priority for ALTH / POSH (OSD, CRSF, LTM)
- resolves #14164
closed 2025-01-17 by haslinghuis