betaflight/betaflight #14168: BF 4.6.0-dev Magnetometer is detected but won't work until barometer is disabled

aramyan, 2025-01-16:
### Describe the bug

On 4.6.0-dev and also on 4.5.1 stable, connecting a Beitian BE-880 GPS module with QMC 5883 magnetometer to GEPRC F405 FC results in magnetometer being detected but showing only zeroes in sensors tab. The FC has a BMP 280 barometer and if that barometer is disabled (by either changing baro_i2c_device from 2 to 0, or setting baro_bustype to NONE, or changing its address to anything other than 0 or 118), the magnetometer starts showing normal values in sensors tab.

This happens both if baro_i2c_address and mag_i2c_address is set to 0, and if they're set to their correct addresses baro_i2c_address = 118, mag_i2c_address = 13. The addresses are not the same, so it's probably not an address conflict.

### To Reproduce

Connect a Beitian BE880 module's SCL and SDA pads to SCL and SDA pads on GEPRC Taker F405 BLS (Target: GEPRC F405) FC, connect 5v and ground pads for power, the magnetometer is detected but shows only zeroes in sensors tab and doesn't react to movements. If barometer is disabled, magnetometer starts sending normal values and reacting to movements.

[image]
[image]
[image]

### Expected behavior

Both barometer and magnetometer should work at the same time.

### Support ID

```txt
5a5d79be-7059-4c94-bc54-401a67687594
```

### Flight controller

GEPRC Taker F405 BLS

### Other components

Beitian BE-880 28x28 GPS module with QMC5883 magnetometer,
GEPRC Maten 1.2g 2w VTX,
GEPRC Nano 900mHz ELRS RX.

### How are the different components wired up (including port information)

GPS TX and RX are wired to RX5 and TX5 pads, GPS works normally. VTX is on UART 1 and ELRS RX is on UART 2.

### Add any other context about the problem that you think might be relevant here

This is a pre-built GEPRC 10", other than adding a BE 880 GPS module, I didn't change anything.

aramyan, 2025-01-23:
UPD: I tried using HGLRC M100-5883 and iFlight Blitz M10 GPS modules, both of which have QMC5883 mag, and I get exactly the same behavior: both I2C devices are detected, but until baro is disabled, mag won't give any readings.

SteveCEvans, 2025-01-28:
Using a MATEKF405TE which exposes I2C bus 1 on pads I can see the baro being accessed at address 0x76 (118) thus:

[image]

With the Beitian BN-880 connected I also see mag access on the same bus at address 0x1e (30) thus:

[image]

Both are detected and working.

[image]

SteveCEvans, 2025-01-28:
Note that this is built using a49960b4 and was also using UART5.

aramyan, 2025-01-29:
Oh, I've had no problem with baro and QMC 5883 mag working on other FCs, an Omnibus F405, HAKRC F405, Aocoda F405 and iFlight F722, it's specifically the GEPRC F405 that can't make them work together at the same time.

SteveCEvans, 2025-01-29:
OK, now I have this test setup I'll see if I can get that FC.

aramyan, 2025-01-29:
Thank you for looking into this, the FC in question is a part of a stack discussed in this article - https://oscarliang.com/geprc-taker-f405/
The exact FC model is GEP-F405-HD V2 if I'm not mistaken.

SteveCEvans, 2025-01-30:
You'll see below that on the MATEKF405TE the default I2C setting sees us running at 800kHz and whilst the rising edge of the SCL is slower than the falling edge as expected, it is quick enough. Top traces are digital, bottom traces are analog and the clock is approximately 800kHz.

[image]

Repeating this with `i2c1_pullup = ON` rather than the default of `i2c1_pullup = OFF` looks no different.

[image]

If we slow the clock down to 100kHz using `i2c1_clockspeed_khz = 100` we get a much cleaner rising edge relative to the pulse width.

[image]

Whilst I suspect the issue here is that the pull-ups are too weak of the GEPRC FC, I'm not able to prove that as I don't yet have one, however if you slow the clock down using `i2c1_clockspeed_khz = 100` you'll mitigate this. Could you please give this a try and report back. It may be that `i2c1_clockspeed_khz = 400` will work equally well.

aramyan, 2025-01-30:
So, I'm getting some weird behavior.

At i2c1_clockspeed_khz = 800, 700 and 600 baro data is received and mag data reads 0-s.
At 500, 400, 300, 200 and 100 mag data is being received and baro data reads 0-s, BUT the board refuses to be recognized when connected to USB half the time. I have to reconnect USB multiple times or physically disconnect mag to get the board to connect to PC and BF.

Also, even when mag data comes in, if i2c1_pullup is OFF, the data would sometimes stop at last received values and not come back until I power cycle. Having i2c1_pullup = ON seems to rectify this, but doesn't affect the other problem.

Also, maybe not related, but this board when connected to USB has 4.9V on all 5V pads and 5.1V on all 9V pads. I don't know if this is normal for this board, but maybe it's just a defective unit.

SteveCEvans, 2025-01-31:
Could you try adding external pull-up resistors from SDA and SCL to 3V3? Try 4K7 first, and if that doesn’t work 2K2.

aramyan, 2025-02-01:
I'll try to find time to do that over the next few days and will report back.

SteveCEvans, 2025-02-17:
I now have the FC. Somewhat confusing that with 4.6 the connector marked with R5/T5 is actually connected to UART2! Anyway, with the baro disabled, the mag is working well. I2C waveforms look good, reaching 3.27V at 800kHz.

[image]

Enabling the baro kills it, and as the baro is on I2C1 and the mag on I2C2 they should be independent.

Hooking up SWD to dig into the software.

aramyan, 2025-02-17:
Thank you, I sadly couldn't find the time to solder pullup resistors, but it seems that it's a problem with the model of FC, not just my unit being defective. Looking forward to solution to this mystery :)

SteveCEvans, 2025-02-18:
OK, so I can now see the point of failure. I've set PINIO 1/2 to the UART1 TX/RX thus:

```
resource PINIO 1 B06
resource PINIO 2 B07
```

I've then wired up to the HD OSD port to drive debug lines using `pinioSet()` for the duration of the baro read on one line and the mag read on the other.

[image]

I then see the following. The mag access fails to complete when the baro access is coincident.

[image]

In more detail:

[image]

So starting one I2C access whilst another is active on another bus is causing issues. I now have an easy trigger condition, so this should be easy to fix.

SteveCEvans, 2025-02-18:
The resistors wouldn't have helped. That was a long shot. You can see from my post above that the rising edge of the SDA/SCL lines are clean which is what matters.

SteveCEvans, 2025-02-19:
I've found the issue. Testing a fix now.

closed 2025-02-22 by haslinghuis