betaflight/betaflight #14182: Mixer mode problem detected

piskaotvaliska31, 2025-01-18:
### Describe the bug

Hello! The situation is this: I decided to upgrade to version 4.5 (before that it was 4.4.x). Everything was fine, I set everything up, and in the motors tab "Mixer mode problem detected", while neither through betaflight nor through ESC config the motors are not spinning. I thought that the ESC firmware was off, I installed different firmware, to no avail. Rolled back to the previous 4.4 and everything worked. Also, the motors do not spin when arming

### To Reproduce

upgrade the Sologood f405 flight controller to version 4.5.X

### Expected behavior

I was counting on the motors to spin through the configurator.

### Support ID

```txt
0cd5f3c6-9ec6-40b6-b5ae-c029867ed496
```

### Flight controller

SoloGood F405 V4 BLS 60A Stack

haslinghuis, 2025-01-18:
Try

```bash
set dshot_bitbang = ON
save
```

piskaotvaliska31, 2025-01-18:
It's working! thank you very much

closed 2025-01-18 by piskaotvaliska31