betaflight/betaflight #14183: 4.5: type mismach (flash.c line 560)

Paku-, 2025-01-20:
### Describe the bug

Cant building 4.5 
error is: type mismach flash.c line 560 vs flash.h

```
flashPartition_t *flashPartitionFindByType(flashPartitionType_e type)
// flashPartition_t *flashPartitionFindByType(uint8_t type)
```

4.6 uses:
`flashPartition_t *flashPartitionFindByType(flashPartitionType_e type)`

### To Reproduce

git pull 4.5 maintenance as of now then:
make clean| make JHEH743_HD

### Expected behavior

clean build :)

### Support ID

```txt
.
```

### Flight controller

JHEH743_HD

Paku-, 2025-01-20:
As it's a newest archlinux I had to:
GCC_REQUIRED_VERSION = 14.2.0

haslinghuis, 2025-01-20:
> GCC_REQUIRED_VERSION = 14.2.0

Firmware 4.6 currently uses 13.3.1 and 4.5 uses 10.3.1.

Just noticed there is a newer version available: https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads but tool chain updates are not back ported.

Paku-, 2025-01-20:
FYI, 4.6 builds perfectly fine with 14.2.0

ledvinap, 2025-01-20:
The code is fixed in master. You can blame it to relevant PR and create backport.

haslinghuis, 2025-01-21:
Fixed in #14186

PR #14186 (2025-01-21): [4.5.2] Fix flash type
- fixes #14183
closed 2025-01-21 by haslinghuis