betaflight/betaflight #14191: Magnetometer issue 4.6

JH3Y50N, 2025-01-22:
### Describe the bug

compass doesn't point north, I've done the calibration several times... I took the gps off the drone to see if it was interference, but it wasn't...

I use walksnail m181 gps compass

I use the connector pointed towards the front of the drone.

In the configuration, I've tried all the configurations with CW and FLIP and it still doesn't point north

I also tried as per walksnail's recommendation

[image]

### To Reproduce

Nani?

### Expected behavior

Point to north

### Support ID

```txt
f4647061-5845-479c-a5bd-788386ddd2ab
```

### Flight controller

DAKE FPV F722

### Add any other context about the problem that you think might be relevant here

[image]

JH3Y50N, 2025-02-22:
Not solved.

haslinghuis, 2025-02-22:
Try setting magnetic declination setting according location

X-Ray-Guy, 2025-03-26:
I am having the same issue with mag alignment, 4.5.1 is fine. But on 4.6 its all over the place no matter of any axis change in orientation choices & mag alignment. I have tried with several GPS / Mag modules. even with goggles on the compass direction in the display is all over the place. My FC alignment is set as was in 4.5.1.
Very Confused....

closed 2025-05-03 by github-actions[bot]