betaflight/betaflight #14200: 4.6 failing to reboot

Paku-, 2025-01-24:
### Describe the bug

4.6 failing on any soft reboot. Hangs on leaving CLI. Cant flash firmware as hangs on attempt to reboot to bootloader.
Yesterday all worked fine. Stacked on given 4.6. Tried to downgrade to 4.5 but the same cant run bootloader so cant flash.

### To Reproduce

JHEH743H >> flash 4.6 then try to reflash.

### Expected behavior

Reflashing on request.

### Support ID

```txt
# support ID
###ERROR IN input: UNKNOWN COMMAND, TRY 'HELP'###
# Support ID
###ERROR IN input: UNKNOWN COMMAND, TRY 'HELP'###
# status
MCU STM32H743 (Rev.V) Clock=480MHz, Vref=3.31V, Core temp=50degC
Stack size: 2048, Stack address: 0x2001fff8
Configuration: CONFIGURED, size: 3579, max available: 131072
Devices detected: SPI:2, I2C:1
Gyros detected: gyro 1, gyro 2 locked dma
GYRO=ICM42688P, ACC=ICM42688P, BARO=DPS310
BUILD KEY: 4cbec0727db738ec9084a0c2f1933328 (4.6.0-dev)
System Uptime: 96 seconds, Current Time: 2025-01-24T23:31:06.699+00:00
CPU:13%, cycle time: 124, GYRO rate: 8064, RX rate: 0, System rate: 10
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 9
FLASH: JEDEC ID=0x00efaa21 128M
GPS: NOT ENABLED
Arming disable flags: RXLOSS CLI MSP
```

### Flight controller

JHEH743HD

Paku-, 2025-01-27:
Have found the define hangs CLI on exit is USE_PWN_OUTPUT. 
Selecting this one is good enough to make app hanging on exiting CLI.

nerdCopter, 2025-02-27:
fixed by https://github.com/betaflight/betaflight/pull/14280 ??

closed 2025-04-06 by github-actions[bot]