betaflight/betaflight #14208: AM32 Configurator can't read ESC

TheIsotopes, 2025-01-26:
### Describe the bug

since #14156 the AM32 Configurator can not read the ESC config

@blckmn pls can you check this

### To Reproduce

use latest master with am32 esc and read out the esc config with AM32 Configurator. you will see that this does not work.

### Expected behavior

I would like to be able to read and write the ESC with AM32 configurator.

### Support ID

```txt
...
```

### Flight controller

TBS Lucid Freestyle

### Other components

TBS Lucid 6S 4in1 esc with latest am32 firmware

blckmn, 2025-01-28:
@TheIsotopes can you please change the motor protocol to ONESHOT and then try again? I just want to confirm it works with PWM motor protocols, and not digital protocols.

rabbitambulance-ra, 2025-01-28:
@blckmn, I know I'm not the original reporter, but I stumbled on this same issue with 4.6 nightly yesterday as well, and can confirm that your hunch is indeed completely correct.

In my case it's a Bayck Dolphin AT32F435 FC, and a couple of different AM32 and BLHeli_S/Bluejay ESCs. 

With the AT32 FC on either 4.5.1 and 4.5.2, both am32.ca (as well as win32 AM32 configurator app) and esc-configurator.com all work as expected with their respective ESCs. On 4.6.x nightly neither am32.ca (or win32 configurator gui) or esc-configurator are able to read ESC settings when set to DSHOT150/300/600.

Switching to ONESHOT42/125 allows both 32-bit and 8-bit ESC configurators to read their respective ESCs even with BF 4.6.x nightly.

I hope this confirmation helps. Thank you very much for the troubleshooting suggestion and for looking at this issue so promptly.

blckmn, 2025-01-28:
Perfect @rabbambulance-ra 

Will have a fix in place tonight.

haslinghuis, 2025-01-28:
@TheIsotopes were you able to test the fix ?

TheIsotopes, 2025-01-28:
@blckmn @haslinghuis I just tested it,
everything seems to be working again 👍 (read / write settings and flashing bootloade / firmware, everything is OK)

TheIsotopes, 2025-01-29:
@blckmn @haslinghuis pls reopen this issue.

Unfortunately, yesterday i was only able to test with an STM32 FC, where everything worked without any problems. 

But today i was able to test again with an AT32 FC and found that it hasn't been fixed.
When i try to read the ESC settings, it seems as if the ESC is constantly being initialized and is stuck in a loop, unfortunately nothing else happens.

haslinghuis, 2025-01-29:
@TheIsotopes can you confirm STM32 still working ?

TheIsotopes, 2025-01-29:
@haslinghuis Yes, i don't see any problems on stm32 fc everything works as it should.

betaflight-admin, 2025-01-29:
> [@blckmn](https://github.com/blckmn) [@haslinghuis](https://github.com/haslinghuis) pls reopen this issue.
>
> Unfortunately, yesterday i was only able to test with an STM32 FC, where everything worked [...]

Will check AT32 now

mituritsyn, 2025-01-29:
> [@haslinghuis](https://github.com/haslinghuis) Yes, i don't see any problems on stm32 fc everything works as it should.

it's not at32 related, code has changed between your tests of stm and at

closed 2025-01-30 by blckmn