betaflight/betaflight #14216: esc serial doesnt compile anymore

KissUltra, 2025-01-29:
### Describe the bug

./src/main/drivers/serial_escserial.c: In function 'serialTimerTxConfigBL':
./src/main/drivers/serial_escserial.c:352:5: error: implicit declaration of function 'TIM_DeInit'; did you mean 'xDMA_DeInit'? [-Werror=implicit-function-declaration]
  352 |     TIM_DeInit(timerHardwarePtr->tim);
      |     ^~~~~~~~~~
      |     xDMA_DeInit
%% (size optimised) ./src/main/drivers/accgyro/accgyro_mpu6050.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_mpu6500.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_spi_bmi160.c 
./src/main/drivers/serial_escserial.c: In function 'onSerialRxPinChangeBL':
./src/main/drivers/serial_escserial.c:389:9: error: implicit declaration of function 'TIM_SetCounter' [-Werror=implicit-function-declaration]
  389 |         TIM_SetCounter(escSerial->txTimerHardware->tim, escSerial->txTimerHardware->tim->ARR / 2);
      |         ^~~~~~~~~~~~~~
./src/main/drivers/serial_escserial.c:389:88: error: 'tmr_type' has no member named 'ARR'
  389 |         TIM_SetCounter(escSerial->txTimerHardware->tim, escSerial->txTimerHardware->tim->ARR / 2);
      |                                                                                        ^~

### To Reproduce

compile with USE_ESCSERIAL

### Expected behavior

Compiles ;)

### Support ID

```txt
6701f88f-26d3-4ac4-8159-e82fe38c157c
```

### Flight controller

4.5.2/4.6.0

nerdCopter, 2025-01-29:
i was able to compile both `4.5-maintenance`@`b72789ee8f` and `master`@`cd84e10fa5` with the make command supplied in the support id.  If there is a different command/config being used, please specify.

<details> 
  <summary>make</summary>
  

`make TBS_LUCID_FC EXTRA_FLAGS="-D'BUILD_KEY=53ccbb223d6f4b46fc0929983c0f06e5' -D'RELEASE_NAME=4.5.2-dev' -DCLOUD_BUILD -DUSE_ACRO_TRAINER -DUSE_DSHOT -DUSE_ESCSERIAL_SIMONK -DUSE_GPS -DUSE_GPS_PLUS_CODES -DUSE_LED_STRIP -DUSE_PINIO -DUSE_SERIAL_4WAY_SK_BOOTLOADER -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_VTX"`
</details>

haslinghuis, 2025-01-29:
Agree this was not related to https://github.com/betaflight/betaflight/commit/cd84e10fa5e75494d8a29330ed488fd32c9810f4

@KissUltra did you rebase and cleanup former builds `git clean -xdf` or `rm -rf obj` ?

KissUltra, 2025-01-30:
***** Commencing Build *****

VERSION:   4.6.0-dev
COMMIT:    cd84e10fa5e75494d8a29330ed488fd32c9810f4
CONTAINER: master
BOARD:     TBS_LUCID_FC/AT32F435M
STARTED:   2025-01-30 07:06:48Z

Want to run this build locally? NOTE: you will need docker installed. See https://github.com/betaflight/cloudbuild for instructions.

> docker run betaflight/cloudbuild:master make TBS_LUCID_FC EXTRA_FLAGS="-D'BUILD_KEY=78920ea4bfdac493143065b0269cecb9' -D'RELEASE_NAME=4.6.0-dev' -DCLOUD_BUILD -DUSE_ACRO_TRAINER -DUSE_DSHOT -DUSE_ESCSERIAL -DUSE_ESCSERIAL_SIMONK -DUSE_GPS -DUSE_GPS_PLUS_CODES -DUSE_LED_STRIP -DUSE_PINIO -DUSE_SERIAL_4WAY_SK_BOOTLOADER -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_VTX"

 -- or (without docker) -- 

> make TBS_LUCID_FC EXTRA_FLAGS="-D'BUILD_KEY=78920ea4bfdac493143065b0269cecb9' -D'RELEASE_NAME=4.6.0-dev' -DCLOUD_BUILD -DUSE_ACRO_TRAINER -DUSE_DSHOT -DUSE_ESCSERIAL -DUSE_ESCSERIAL_SIMONK -DUSE_GPS -DUSE_GPS_PLUS_CODES -DUSE_LED_STRIP -DUSE_PINIO -DUSE_SERIAL_4WAY_SK_BOOTLOADER -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_VTX"

***** Docker execution log: *****

Pulling from Github (cd84e10fa5e75494d8a29330ed488fd32c9810f4):
From https://github.com/betaflight/betaflight
 * [new ref]             cd84e10fa5e75494d8a29330ed488fd32c9810f4 -> CLOUD_HEAD

Checking out the branch -> CLOUD_HEAD (for cd84e10fa5e75494d8a29330ed488fd32c9810f4)
Previous HEAD position was ac82d8b99 Refactoring of PR 13050 - support for custom OSD messages from external device (#14097)
Switched to branch 'CLOUD_HEAD'

Commencing the build (make TBS_LUCID_FC):
git submodule update --init --remote -- src/config
Submodule 'config' (https://github.com/betaflight/config.git) registered for path 'src/config'
From https://github.com/betaflight/config
   16ac022..5e33fe4  master     -> origin/master
Submodule path 'src/config': checked out '5e33fe422e2c1fb095db84508a9e3ebb465a80a4'
Building target config TBS_LUCID_FC
make -j hex CONFIG=TBS_LUCID_FC
make[1]: Entering directory '/source'
make  ./obj/betaflight_4.6.0_AT32F435M_TBS_LUCID_FC.hex
make[2]: Entering directory '/source'
EF HASH -> ./obj/main/AT32F435M_TBS_LUCID_FC/.efhash_02e3ca22d2891a2a04b305038140ed03
%% startup_at32f435_437.s 
%% (speed optimised) ./src/platform/common/stm32/system.c 
%% (optimised) ./src/platform/common/stm32/io_impl.c 
%% (optimised) ./src/platform/common/stm32/config_flash.c 
%% (optimised) ./src/platform/AT32/startup/at32f435_437_clock.c 
%% (optimised) ./src/platform/AT32/startup/system_at32f435_437.c 
%% (optimised) ./src/platform/AT32/adc_at32f43x.c 
%% (optimised) ./src/platform/AT32/bus_i2c_atbsp.c 
%% (optimised) ./src/platform/AT32/bus_i2c_atbsp_init.c 
%% (optimised) ./src/platform/AT32/bus_spi_at32bsp.c 
%% (optimised) ./src/platform/AT32/camera_control_at32.c 
%% (optimised) ./src/platform/AT32/debug.c 
%% (optimised) ./src/platform/AT32/dma_at32f43x.c 
%% (optimised) ./src/platform/AT32/dma_reqmap_mcu.c 
%% (optimised) ./src/platform/AT32/dshot_bitbang.c 
%% (optimised) ./src/platform/AT32/dshot_bitbang_stdperiph.c 
%% (optimised) ./src/platform/AT32/exti_at32.c 
%% (optimised) ./src/platform/AT32/io_at32.c 
%% (optimised) ./src/platform/AT32/light_ws2811strip_at32f43x.c 
%% (optimised) ./src/platform/AT32/persistent_at32bsp.c 
%% (optimised) ./src/platform/AT32/pwm_output_at32bsp.c 
%% (optimised) ./src/platform/AT32/pwm_output_dshot.c 
%% (optimised) ./src/platform/AT32/rcc_at32.c 
%% (optimised) ./src/platform/AT32/serial_uart_at32bsp.c 
%% (optimised) ./src/platform/AT32/serial_uart_at32f43x.c 
%% (optimised) ./src/platform/AT32/serial_usb_vcp_at32f4.c 
%% (optimised) ./src/platform/AT32/system_at32f43x.c 
%% (optimised) ./src/platform/AT32/timer_at32bsp.c 
%% (optimised) ./src/platform/AT32/timer_at32f43x.c 
%% (optimised) ./src/platform/AT32/usb_msc_at32f43x.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_mpu.c 
%% (optimised) ./src/main/drivers/dshot_bitbang_decode.c 
[... 71 lines cut ...]
      |         ^~~~~~~~~~~~~~
./src/main/drivers/serial_escserial.c:383:88: error: 'tmr_type' has no member named 'ARR'
  383 |         TIM_SetCounter(escSerial->txTimerHardware->tim, escSerial->txTimerHardware->tim->ARR / 2);
      |                                                                                        ^~
%% (size optimised) ./src/main/config/config_eeprom.c 
%% (size optimised) ./src/main/config/config_streamer.c 
%% (size optimised) ./src/main/config/feature.c 
%% (size optimised) ./src/main/config/simplified_tuning.c 
cc1: all warnings being treated as errors
%% (size optimised) ./src/main/cli/cli.c 
%% (size optimised) ./src/main/cli/settings.c 
make[2]: *** [Makefile:488: obj/main/AT32F435M_TBS_LUCID_FC/drivers/serial_escserial.o] Error 1
make[2]: *** Waiting for unfinished jobs....
%% (optimised) ./src/main/drivers/dshot.c 
make[2]: Leaving directory '/source'
make[1]: *** [Makefile:611: hex] Error 2
make[1]: Leaving directory '/source'
make: *** [mk/config.mk:72: TBS_LUCID_FC] Error 2
make failed with 2

 ***** Completed Build *****

Build time: 2025-01-30 07:06:54Z, time taken: 7.25

Submitted:  2025-01-30 07:06:45Z, wait time: 9.25

KissUltra, 2025-01-30:
tried to compile this morning.... no joy.

KissUltra, 2025-01-30:
Its a cloud build.

haslinghuis, 2025-01-30:
`USE_ESCSERIAL` does not compile with AT32 (even when added to target.h)

KissUltra, 2025-01-30:
This is MUST HAVE feature for voltara esc.

haslinghuis, 2025-01-31:
https://build.betaflight.com/api/builds/a60edf56744638874b40984ced65527c/log

haslinghuis, 2025-01-31:
@KissUltra see https://build.betaflight.com/api/builds/c256dafa1815d97ca19708f13cfac097/log

Please test #14228

closed 2025-03-28 by blckmn