betaflight/betaflight #14217: Arming copter fails after more than 20 minutes of waiting

DJ-Von, 2025-01-29:
### Describe the bug

After starting the copter and waiting for more than 20 minutes (I couldn't determine the exact time) I couldn't arm it. I looked at the code, debugged using serial_printf, and found out that when trying to arm after more than 20 minutes, the program enters this condition in the `tryArm` method in the `src/main/fc/core.c` file: https://github.com/betaflight/betaflight/blob/cd84e10fa5e75494d8a29330ed488fd32c9810f4/src/main/fc/core.c#L535-L548

Moreover, the program enters only the first condition, but it does not enter any of the subsequent ones.
I couldn't figure out why this is happening.

### To Reproduce

It is necessary to supply power to the copter and wait 20-30 minutes.

### Expected behavior

You won't be able to arm

### Support ID

```txt
9c1cb37c-96dd-4914-95b5-5d4d5e0b4846
```

### Flight controller

STM32F405RGT6

### Other components

MCU: STM32F405RGT6
target: AOCODARCF405V3

Quick-Flash, 2025-01-29:
Seems like the cmpTimeUs(currentTimeUs, getLastDshotBeaconCommandTimeUs()) is overflowing and not handling that correctly.

ledvinap, 2025-01-29:
Actually, the minimum delay is likely 36 minutes (2^31us). There will be multiple cases of this.

haslinghuis, 2025-01-29:
But https://github.com/betaflight/betaflight/blob/cd84e10fa5e75494d8a29330ed488fd32c9810f4/src/main/io/beeper.c#L453-L454
should reset it

closed 2025-01-31 by DJ-Von

ledvinap, 2025-01-31:
@DJ-Von : Be careful when using BF for over 36 minutes. Some bugs may be triggered

DJ-Von, 2025-01-31:
@ledvinap:  What could be the reason for this?

ledvinap, 2025-01-31:
The same problem as you found, just in different place. There is also negligible chance that `micros()` will return `0` after 72  minutes, which will break assumptions in a lot of places, but that probability is  hopefully very small