betaflight/betaflight #14219: Memory bug in MSP with custom OSD messages

jefffhaynes, 2025-01-29:
[This line](https://github.com/betaflight/betaflight/blob/cd84e10fa5e75494d8a29330ed488fd32c9810f4/src/main/msp/msp.c#L4150) appears to incorrectly use ```strlen``` in an effort to grab the string length for the incoming MSP custom message value.  However, the string is prefixed with the length, **not** null terminated.  Even if the string was meant to be null-terminated, this is unsafe and results in a seg fault when the incoming buffer does not contain a null here (which it will likely not).

Instead ```memset``` should use the string length, which is already available here as ```textLength```.

ledvinap, 2025-01-29:
it is a bug, everything until first \0 is zeroed, possibly overwriting data after text buffer

jefffhaynes, 2025-01-29:
Ok, yeah I don't know enough about the inner workings of betaflight to know how that data is structured nuts definitely suspicious and I'm definitely getting set faults!  Thanks for looking at it.

ledvinap, 2025-01-29:
can you create PR to fix it, please? 
textSpace is correct length for zeroing

jefffhaynes, 2025-01-29:
Sure np

jefffhaynes, 2025-01-29:
Is it sufficient to zero the text length?  Shouldn't the copied (read) string be null terminated?

commit 79ecc1ac (2025-02-01): Fix MSP memory issue (#14221)

* fixes #14219

* Update src/main/msp/msp.c

---------
closed 2025-02-01 by haslinghuis