betaflight/betaflight #14225: After the update 4.5.1 HGLRCF411: index not between 1 and 10

gumileeev, 2025-01-31:
### Is your feature request related to a problem? Please describe

I upgraded from 4.4.3 to 4.5.1, after trying to restore settings - I get the following error:
# resource SERIAL_TX 11 B03
###ERRROR IN resource: INDEX NOT BETWEEN 1 AND 10###

At the same time, after the update, the GPS light in the application stops lighting up. My GPS is connected to the place where the manufacturer originally intended the SpeedyBee WiFi to be connected. If I return to 4.4.3, everything works again, but I really want 4.5.1

I don't understand what I should do to fix the error?

### Describe the solution you'd like

I don't understand why this error occurs and how I should fix it.

### Describe alternatives you've considered

I don't know any other alternatives.

### Other information

I am attaching the saved settings from 4.4.3 and what happens after restoring the settings to 4.5.1

[HGLRCF411 4.4.3.txt](https://github.com/user-attachments/files/18621313/HGLRCF411.4.4.3.txt)
[HGLRCF411 4.5.1.txt](https://github.com/user-attachments/files/18621314/HGLRCF411.4.5.1.txt)

haslinghuis, 2025-01-31:
- add `SOFTSERIAL` to Other Options and flash again
- Use `resource SOFTSERIAL_TX 1 B03` instead

See also https://betaflight.com/docs/wiki/guides/current/SoftSerial

gumileeev, 2025-01-31:
> * add `SOFTSERIAL` to Other Options and flash again
> * Use `resource SOFTSERIAL_TX 1 B03` instead
> 
> See also https://betaflight.com/docs/wiki/guides/current/SoftSerial

Thank you! I made these settings, after that I got the GPS icon.
I still have problems:
1. The OSD on the screen is updated literally line by line, from left to right, from top to bottom, for the screen to update - about 10 seconds should pass.
2. When connecting to the computer, I see the LOAD MSP flag. When I disconnect from the computer and try to make an arm - it writes the error "LOAD" and beeps, does not start the motors.

haslinghuis, 2025-01-31:
- Try 4K pid loop and dshot 300 motor protocol.

nerdCopter, 2025-01-31:
just a note: `SOFTSERIAL` custom define seems to only work in 4.6.  when building 4.5 it errors due to redefine. (when building `HGLRCF411`)

gumileeev, 2025-01-31:
> * Try 4K pid loop and dshot 300 motor protocol.

Thank you very much, now I can fly!

closed 2025-01-31 by gumileeev