betaflight/betaflight #14226: Loosing sensors on H743

Pancronos, 2025-01-31:
### Describe the bug

I had correctly set rangefinder/optical flow, today i connected again and sonar icon was off. I haven't done any change. This is an H743 but it's the MICOAIR AIO, looking at pinout there are differences, could this be the problem?
H743
https://micoair.com/flightcontroller_micoair743/
H743 AIO
https://micoair.com/flightcontroller_micoair743_aio_35a/

[image]

Backup that is ok

[BTFL_cli_backup_CINEBOT30MICO_20250127_194904_MICOAIR743.txt](https://github.com/user-attachments/files/18622240/BTFL_cli_backup_CINEBOT30MICO_20250127_194904_MICOAIR743.txt)

After few days (no change)

[BTFL_cli_backup_CINEBOT30MICO_20250131_182205_MICOAIR743.txt](https://github.com/user-attachments/files/18622243/BTFL_cli_backup_CINEBOT30MICO_20250131_182205_MICOAIR743.txt)

### To Reproduce

set rangefinder_hardware = MTF01
set opticalflow_hardware = MT
save 

and after sometime reconnect

### Expected behavior

sensor setting should not change

### Support ID

```txt
Id: "f99369e3-309b-4726-902d-0ebdea52c4f7"
```

### Flight controller

4.6-dev

### Other components

MTF01 rangefinder on uart 4

haslinghuis, 2025-01-31:
```
The flight controller part of MicoAir743-AIO is almost the same as  MicoAir743, except for:

    Change M9(TIM4->CH3) to LED Strip control pin
    Change M10(TIM4->CH4) to Beep- control pin
    Delete IST8310 (on board compass)
    Delete AT7456E (on board osd)

MicoAir743-AIO can share the same firmware with MicoAir743, unless LED strip control and beeper are required.
```

Pancronos, 2025-01-31:
After adding:
set rangefinder_hardware = MTF01
set opticalflow_hardware = MT
save
and modifying gps angle > save  again result in corrupted config 
set opticalflow_hardware = 
###ERROR IN diff: CORRUPTED CONFIG: opticalflow_hardware = 224###

haslinghuis, 2025-01-31:
gps angle ???

[image]

haslinghuis, 2025-01-31:
Verify using MicoAssistant module is still set to MSP ?

Pancronos, 2025-01-31:
I don't think is an MSP issue, also all the MTF i have were factory set for inav (msp). It looks like more as a conflict.
 Anyway will check and report

haslinghuis, 2025-01-31:
Yes just need a way to reproduce the issue :)

Pancronos, 2025-01-31:
read result of MTF01:

[image]

haslinghuis, 2025-01-31:
Is this build local?

```bash
# version
# Betaflight / STM32H743 (SH74) 4.6.0 Jan 27 2025 / 18:39:11 (norevision) MSP API: 1.47
# config rev: 578caf0
```

haslinghuis, 2025-01-31:
Please try again on latest development build as lots of changes were made recently.

Pancronos, 2025-01-31:
Build log 
https://build.betaflight.com/api/builds/fc43a9217c88fb6080e725cd50e58e7e/log

Backup before flashing

[BTFL_cli_backup_CINEBOT30MICO_20250131_210705_MICOAIR743.txt](https://github.com/user-attachments/files/18623903/BTFL_cli_backup_CINEBOT30MICO_20250131_210705_MICOAIR743.txt)

Afteer flashing was ok, decreased gps yaw 1 degree:

[BTFL_cli_backup_CINEBOT30MICO_20250131_211022_MICOAIR743.txt](https://github.com/user-attachments/files/18623909/BTFL_cli_backup_CINEBOT30MICO_20250131_211022_MICOAIR743.txt)

Pancronos, 2025-01-31:
I did few other tests, disabling baro, changing arming angle or any change on GPS, erase rangefinder/opticalflow hardware
It seems that any change that reboot FC creates the problem.

Pancronos, 2025-02-01:
Unfortunately no change for me, triggerring baro or changing gps values brake config
set opticalflow_hardware = 
###ERROR IN diff: CORRUPTED CONFIG: opticalflow_hardware = 224###

tested with firmware version
https://build.betaflight.com/api/builds/3b24bb593fbb50dbfdac50dfe7360f7d/log

Pancronos, 2025-02-01:
I confirm it finally all works!! Good job!!

closed 2025-02-01 by Pancronos